gawindx/WinNUT-Client #4: Re-establish connection automatically

asitnik, 2019-03-21:
Greetings, first thanks for this project, it is great and I will recommend it to my colleagues.
Although the "Re-establish connection" checkbox gets marked on the "Preferences --> Connection" menu, the connection does not get automatically re-established (and the checkbox mark disappears). This happens whether or not the program is run as an administrator.
Thanks for your project and attention.

tobsibobsi, 2019-03-27:
yes it's definitely a nice project. Good job!
But like the issue-opener said, I'd would be great if that bug "Re-establish connection" on Windows 10 would work again. My UPS is connected via USB to the NAS. Sometimes,somehow the connection between UPS and NAS get lost, and come up again pithing seconds, but WinNutClient does neither re establish the connection, nor saves the ticked checkbox.
Thanks for that great project. May you can fix this issue for Windows 10.

asitnik, 2019-03-28:
Just to clarify, the server is a Raspberry Pi-2 and the client losing connection is Windows 10. I had tried using my NAS as server, and the connectivity with the NAS was unstable: every time the NAS restarted, the USB device ID changed; but with the Pi it stays the same.

gawindx, 2019-04-01:
Hello and thank you for your support.
I currently have a lot of work and I will not be able to provide a solution in the next few days.

I do not think that the problem is related to Windows 10 but rather to the fact that it is a part of the code that I have not yet appropriated myself and that the value of the checkbox is simply not used.

Version 17.1 is under development and will provide the ability to translate the interface.
I will make the necessary modifications to an automatic reconnection on this same version.

I think the function will be limited to an attempt every 30 seconds for 15 min - beyond this waiting time, a custom error / icon can be displayed and the reconnection will be manual.

asitnik, 2019-04-02:
Thanks for replying, Indeed it is not a Windows issue; every time I restart the Pi server the connection drops and does not get automatically re-established. It is not a Pi issue either, as it properly restarts and correctly detects the USB device, so it is just a missing feature (not a bug) for which all of us will be thankful when it gets implemented. Once again, let's express our appreciation for your good work!

tobsibobsi, 2019-04-02:
Thank you very much so far.
we are looking forward for the next Version to get rid of the not re-established connections after connection-loss

gawindx, 2019-04-15:
Things promised, things due

I published the new version supporting multi language and correcting the problem of reconnection (normally).
I am waiting for your feedback on this point because on my side I have no problem of inadvertent disconnection.
I proceeded by simulation but nothing is worth the real tests.

asitnik, 2019-04-15:
Working well here; I restarted my Raspberry Pi to test, and the connection came back after the Pi restarted the services.
Thank you so much for this feature!

gawindx, 2019-04-16:
Thanks for your feedback.
I think we can now consider that the problem is solved.

closed 2019-04-16 by gawindx

tobsibobsi, 2019-04-16:
Windows defender says Bearfoot.a ml 
Trojan 🥺 
No more use for that software.

asitnik, 2019-04-16:
> Windows defender says Bearfoot.a ml
> Trojan 🥺
> No more use for that software.

Run it through virustotal.com, you'll see it's clean. Being a program that does a network probe of sorts, it does look malicious to some detection engines' behaviour analysis.
https://www.virustotal.com/gui/file/332ab03503bc87599aa5eff30cb7c0d845dd69c632893b96d63cd260a9d059e8/detection

gawindx, 2019-04-17:
First of all, thanks to Asitnik for anticipating my answer and indeed, this is a false positive.

I advise you to configure an exclusion on the executable itself or its directory to avoid any inappropriate report from Windows Defender (or other).

In a second time, I would thank you for not polluting unnecessarily a closed subject and to create your own issue; I will answer them with the same care.