gawindx/WinNUT-Client #7: reported as malware by avast

yo8aiv, 2019-05-10:
hmm ... reported as malware by avast !

gawindx, 2019-05-16:
Answer already made.

It's a false positive

Check out https://github.com/gawindx/WinNUT-Client/issues/4#issuecomment-483802813 and https://www.virustotal.com/gui/file/332ab03503bc87599aa5eff30cb7c0d845dd69c632893b96d63cd260a9d059e8/detection

yo8aiv, 2019-05-19:
ok. added exception. using it. 
pretty good .thx. 
i have a few suggestions: if possible 
- to add few screenshots to your wiki here .. 
- i dont think calibration is working corectly on analog dials.
- curved scale for analog dials (with calibration posibility 0-30V)
- add digital dials choice
- increase window size and text PLEASE :)

Thanks in advance. Nice job.

Boia11, 2019-07-18:
@yo8aiv The problem is inside Autoit implementation, read the comment here: [AutoIt_Malware](https://www.virustotal.com/gui/file/332ab03503bc87599aa5eff30cb7c0d845dd69c632893b96d63cd260a9d059e8/community)

funkybunch, 2019-09-16:
That Virus Total link must have been updated.  Still getting reported as Malware and according to that link as it stands now, looks far from safe.  I'm going to look for alternative solutions, and until this is resolved in some form I encourage others coming across this to do the same.