gawindx/WinNUT-Client #12: executable is reported to have virus on win10

richo67, 2020-05-04:
the executable of release of 1.7.2 is reported to contain  a virus by Win10.

gawindx, 2020-06-03:
This is a false positive due to the AutoIt compiler.
This does not only affect my project but all the scripts compiled via AutoIt.
Look here :
https://www.autoitscript.com/wiki/AutoIt_and_Malware

closed 2020-06-03 by gawindx

richo67, 2020-06-05:
Well, it may be so but I am not able to install it as the antivirus is removing it immediately and I have, unfortunately no control over it. So I am using older version which has not the issue. If all your project are build the same way does it counts for older build too?

gawindx, 2020-06-05:
All versions are compiled in the same way, so you risk having the same problem regardless of the version. I only see 3 possibilities:
 - create an exclusion at the antivirus level
 - declare the executable with the editor of your antivirus for analysis, in this way they will be able to validate it and it will no longer be recognized as bad. The problem is that it will have to be renewed with each new version because the signature of the executable will change
 - install autoit (I think Autoit signature is already in white list), download the sources of WinNut and execute the script upsclient.au3

richo67, 2020-06-05:
Nicolas,
thank you for the suggestion and I may try to utilize them, but it is Win10 itself and its malware protection which does remove it.

gawindx, 2020-06-05:
For WinDefender, you will find the procedure below (the simplest is to create an exclusion on the WinNut directory rather than on the executable itself):

https://support.microsoft.com/en-us/help/4028485/windows-10-add-an-exclusion-to-windows-security

richo67, 2020-06-05:
Thank you very much for your kind help Nicolas.

jhujasonw, 2020-06-17:
Just one other comment, if you download autoit and make the binary yourself from the source, Windows Defender doesn't flag it.  I did this and manually scanned the file.  Comes back clean.

gawindx, 2020-06-23:
In my case, WinDefender continues to recognize the executable as a virus / malware, even compiled from my machine.

I have set up a program submission to VirusTotal and I am responsible for reporting false positives to antivirus vendors in order to resolve this problem.
I will see to use but I am afraid of having to submit each new version this way.

Unless you rewrite the whole project in Visual Studio / C ++ ... which I don't really like

richo67, 2020-07-08:
This exclusion didn't help.
I can not run it. Even when I stopped runtime protection in Win10 it could be run but after 2-3 seconds it was killed and removed from folder which was in exclusion from which I run it.
-r-

> 
> 
> For WinDefender, you will find the procedure below (the simplest is to create an exclusion on the WinNut directory rather than on the executable itself):
> 
> https://support.microsoft.com/en-us/help/4028485/windows-10-add-an-exclusion-to-windows-security

gawindx, 2020-07-08:
Sorry that couldn't help you anymore.

I assure you that the program is healthy but I take good account of the extent of this problem.

The initial project not having been developed by myself, I only took it over and improved it but I realize that the choice of Autoit, which was perhaps wise at the time, was no longer an interesting choice.

I undertook to rewrite the whole project under Visual Basic, which should make it possible to no longer be bothered by these false positives.

For now, I would say that the progress of the code is close to half (or even a little more) and that I should be able to release V2 in less than a month.
Another advantage of this rewriting, it should allow an executable consuming less resource because at the moment, it consumes less than 10Mo of Ram against 20-30Mo for the Autoit version.

At the level of the advancement of the code, I still have the implementation / improvement of certain GUIs and in major point, the design of the dials which remain very important and for which I am still looking for documentation.