gawindx/WinNUT-Client #13: IPv6 support

asitnik, 2020-05-24:
Is it possible to have it support IPv6? Looks like it only supports IPv4 now. Thanks.

gawindx, 2020-06-03:
Unfortunately, it seems that AutoIt does not support IPV6, so it is not possible for me to add this functionality.

gawindx, 2020-06-04:
after some research, there may be a solution with the netsh command. the portproxy function would create a TCP proxy between an IPV4 address to an IPV6 address. I can not test on my side but if you can validate the operation, I think I can incorporate it into the project transparently. launch the following command (modified according to your needs):

`netsh interface portproxy add v4tov6 listenaddress=localhost listenport=local_port connectport=nut_port_server connectaddress=ipv6_nut_server `

it may be necessary to launch command via PowerShell. then, in the WinNut Options, indicates localhost (or the address "listenadress") for the ip and the port "listenport" for the port.

to delete the portproxy, simply enter the following command:

`netsh interface portproxy delete v4tov6 listenaddress=localhost listenport=local_port`

keep me posted and if it works i would modify the code accordingly

gawindx, 2020-06-05:
@asitnik 
After testing an IPV4 to IPV4 proxy port, it seems to work.
So I will see to implement this functionality in a transparent and automatic way to make it compatible with IPV6.

gawindx, 2020-06-09:
I have just released version 1.7.3.0 which contains several updates.
Among other things, I made WinNut compatible with IPV6, although AutoIt is not.

These modifications use the netsh portproxy command which requires administrator rights to run (incorporated in the code so you just have to accept at startup).

I'm waiting for your return to find out if it works.

If everything is ok on your side, I will close the request.

gawindx, 2020-06-23:
I am currently releasing version 1.8 which contains several updates including support for IPV6.

Make me a return to tell me if it works on your side (for my part, I simulated the function with a portproxy IPV4-> IPV4 and it worked without problem, but I do not work in IPV6 therefore ...)

I close this issue but do not hesitate to tell me if something is not working.

closed 2020-06-23 by gawindx

gawindx, 2020-07-18:
I released a beta version under visual basic to fix the problem of false positives.

I would like to have your feedback on its compatibility in IPV6.
Can you test it and tell me if it works?

You can find her [here](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.1.0-dev)

Thank you