gawindx/WinNUT-Client #14: Wi-fi support

asitnik, 2020-05-24:
Greetings,
I'd like to have the client connect to the server via wi-fi, and it doesn't work; it requires an Ethernet connection. Can that be done? Thanks.

gawindx, 2020-06-03:
I am quite surprised by your problem.
The connection is made via IP, and therefore it has nothing to do with wi-fi (unless it has to do with your question about IPV6).

gawindx, 2020-06-23:
Keep me informed if the addition of IPV6 support solved your WiFi problem or if it is still current.

While waiting for your return, I leave this issue open.

asitnik, 2020-06-23:
It looks good, thanks. Version 1.8 looks awesome!

Ari Sitnik
justaskenterprises@gmail.com
Cell: 647-274-9665
LinkedIn: ca.linkedin.com/in/asitnik
Twitter: twitter.com/asitnik
Skype: asitnik <https://is.gd/qKKvAr>

asitnik, 2020-07-29:
I've skipped over to 2.0.x and, so far, so good.

gawindx, 2020-08-08:
I think that with V2, this problem is no longer relevant

I close the issue, it will have to be reopened if necessary

closed 2020-08-08 by gawindx