gawindx/WinNUT-Client #17: bug ? and proposal

testlog0, 2020-06-13:
First of all, a big thank you for this application, it allows me to view on a remote PC the status of my UPS (BR1500G-FR) connected to my nas (DS218+).

Small problem: the output voltage remains at 0, a bug?

The following, I don't know if it's achievable but I still propose (you never know):
is it possible to just display in the "UPS load" part the wattage in addition to the percentage?
Is it possible to export the data to a file? (For example, at each updating, export the updated values in a TXT file or other.)

Again thank you for this interface.

PS: sorry for the mistakes, I'm not good at English.

gawindx, 2020-06-13:
first of all, thank you for this return.

Regarding the problem you are encountering as well as your request for improvement, I am currently working on version 1.8 which brings several improvements as well graphics as on the possibility of having a log file, which will allow to have more information in a case like yours.

I will therefore not immediately meet your expectations but I will inform you when it is made available in order to have more information to deal with this potential bug.

I think I can make this new version available within 1 to 2 weeks.

Regarding version 3.7.1, you are talking about the version of nut or WinNUT, because for the moment the version of WinNUT is in 1.7.3.1

testlog0, 2020-06-13:
Yes, 1.7.3.1 (sorry 😄)
Thanks in advance for the next version. 💪

gawindx, 2020-06-17:
I just released a pre-release of version 1.8.
This pre-release is in version 1.7.9.9 but contains the fully functional logging functionalities.

You will see new options in Preferences-> Miscellaneous regarding the activation of logging as well as the level of information.

To solve your problem, put yourself at Debug level, the file will be created and accessible as soon as possible.

I advise you not to let it run too long at this level because the log file will grow very quickly and this will also generate a slight slowdown in WinNut due to regular disk access.

Once the problem is found, deactivate the logs and make the file available to me ([https://www.uploadedit.com/_to-upload-documents-onto-internet-PLAIN-TEXT-TXT-hosting.htm](url) or other service) .
I could analyze it and try to understand what is going on with your inverter (by the way, it will serve as a test to validate the relevance of the logging function).

This version also incorporates a modification at the level of the icons as well as "dynamic" icons which reflects the state of the inverter / battery in real time (the icons being small, it remains succinct).

You can download the exe in the Dev branch in the Build directory or directly by this link :
[https://github.com/gawindx/WinNUT-Client/blob/Dev/Build/upsclient.exe](url)

Ririx02, 2020-06-17:
I tested this version and here are the logs:
http://m.uploadedit.com/busd/1592418731541.txt
it seems that it is the value which is not sent by the NUT synology or that it cannot read it? (GetUPSData output-> voltage Return -1)
I also noticed that the frequency also had a problem too: GetUPSData input-> frequency Return -1

Ririx02, 2020-06-17:
and actually, with the log activated, I have the display and the logs that are done correctly but I can neither access the options (so I stopped the logs directly from the configuration file) nor stop the program ( I turned it off with an "end of task" in the task manager)

gawindx, 2020-06-17:
like i said, this is a pre version.
I have no worries on my pc if it is not a slight slowdown when I put myself in debug.

your return lets me see that it is not perfectly developed.
I think I will still make some modifications to improve this.

I'm going to study your logs from tomorrow and try to see how we can get around your problem. for the frequency, there is an option to choose 50 or 60 Hz (since version 1.7.3.1); tell me if you defined it or not.

if it is defined it is that there is a bug at this level in my code. if it is not defined, try to activate it and see if it is better in which case I would reproduce the same mechanism for the tension which causes you problem.

Ririx02, 2020-06-17:
For frequency, it is set to 50.
(Little funny thing: when I test to set it to 60hz, the graph tells me that the frequency is at 220Hz 🤣)

gawindx, 2020-06-17:
actually, it's surprising

I looked at the logs a bit (on the phone it's not very clear) I can see that your inverter (or the nut driver) also does not provide the battery capacity.
and for the output voltage, it should normally be equal to the input voltage when nut does not return it.

I'm going to review that too.

I think the 1.8 will take a little longer to come out than expected ....

Ririx02, 2020-06-17:
Currently, to view the info from my UPS, I go through the NUT server of my Synology NAS. Could the problem come from Synology's NUT service?

Ririx02, 2020-06-17:
It is a DS218 + (DSM 6.2.3-25426)

gawindx, 2020-06-17:
I do not know this nas, but logically, if you see this information from its interface, there is no reason that we can not recover it.

on the other hand, if the nas does not give you these missing information, there is a good chance that we cannot obtain them.

Ririx02, 2020-06-17:
From the interface, there is even less information than on your application.
On the Synology interface, it only gives the manufacturer, model, status, battery charge and remaining time.
But apparently Synology is using Network UPS Tools.
https://roll.urown.net/NAS/nas-ups.html
(I hope his info can help you)

gawindx, 2020-06-18:
I don't know if you have access to a linux console on your nas, but if it is, you should try to run the following command:
  `upsc name_of_your_ups` 
(name defined in the settings of WinNut)

it should return something like this:
`Init SSL without certificate database`
`battery.capacity: 7`
`battery.charge: 255`
`battery.runtime: 3932100`
`battery.voltage: 13.6`
`battery.voltage.nominal: 12`
`device.mfr: RPS S.p.a.`
`device.model: USV5`
`device.serial:`
`device.type: ups`
`driver.name: riello_usb`
`driver.parameter.pollinterval: 2`
`driver.parameter.port: auto`
`driver.parameter.synchronous: no`
`driver.version: 2.7.4`
`driver.version.internal: 0.03`
`input.bypass.frequency: 409.50`
`input.bypass.voltage: 4095`
`input.frequency: 50.10`
`input.voltage: 238`
`output.frequency: 50.10`
`output.frequency.nominal: 50.0`
`output.L1.current: 0`
`output.L1.power: 0`
`output.L1.realpower: 0`
`output.L2.current: 0`
`output.L2.power: 0`
`output.L2.realpower: 0`
`output.L3.current: 0`
`output.L3.power: 0`
`output.L3.realpower: 0`
`output.power.percent: 23`
`output.voltage: 234`
`output.voltage.nominal: 230`
`ups.firmware: SWM036-01-00`
`ups.load: 23`
`ups.mfr: RPS S.p.a.`
`ups.model: USV5`
`ups.power.nominal: 600`
`ups.productid: 5500`
`ups.realpower.nominal: 360`
`ups.serial:`
`ups.status: OL`
`ups.temperature: 255`
`ups.vendorid: 04b4`
this will allow you to know the variables supported by the driver nut for your inverter.

If the desired info is on another parameter than the one to query, I could adapt the code to test a second value in the case where the first returned is equal to -1.

On the other hand, if the information is not returned, we will know that the pilot is either, not up to date, or incomplete.

gawindx, 2020-06-18:
Well, I think at least it's clear.
Unfortunately, we will not be able to do much.

Either the driver for this inverter is old and there is a newer one (but here I am far from being an expert on this subject) or we will have to do otherwise.

What I can do at my level is to transfer the input voltage to the output voltage as a fallback value but I could not invent a value that does not exist.

gawindx, 2020-06-18:
I just pushed a new version of the exe (by the way, the logs should slow down much less).

Try and tell me if now you have a value that shows up.
It is a fallback value, namely the value of the input voltage transferred to the output but in your case, I can not do much better.

You will find it on the following link (the exe is no longer called upsclient.exe but WinNut-client.exe) :
[https://github.com/gawindx/WinNUT-Client/blob/Dev/Build/WinNut-client.exe](url)

gawindx, 2020-06-18:
Next, there would be a more recent version of the driver (USB HID 0.95), this would be version 0.96 but I don't know where to find it yet.

gawindx, 2020-06-18:
I think I have corrected the problem and that it should show you a voltage (equal to the input voltage).
The updated executable is available through the same link.

That said, it just remains visual because if your inverter is offline, you will see an output voltage at 0V ....

Regarding version 0.96, it dates from 2015 ... And it is part of nut directly.
This means that the version of nut included in the firmware of your nas is simply a version of at least 5 years.

My version:
`driver.version: 2.7.4`

Your version:
`driver.version: DSM6-2-25364-191230`

Note: version 2.7.4 is the latest in nut and it already dates from 2016

I think synology has a version 2.7.3 or 2.7.2

Maybe there are tutorials on the internet to update it but just like you, I would think twice before doing it because I have no idea what impact it might have on how your NAs work

Ririx02, 2020-06-18:
Yes, I think I will wait if, one day, they make an update for the nas because I do not want to make it unusable. 
Thank you for this version, even if it is not the right value, it is always more beautiful than a graph at 0. 😋

In any case, a big thank you for taking the time to look at this problem with me.👍

gawindx, 2020-06-23:
I am publishing the final version 1.8 which contains the latest updates, including the display of instantaneous power at the battery level.

I am going to close this issue but don't hesitate to let me know if something is not working properly.

closed 2020-06-23 by gawindx