gawindx/WinNUT-Client #19: Round the value of watts

Ririx02, 2020-06-25:
Hello,
It's me again. 😁
Just a quick question, would it be possible to round the value "UPS load" in Watt to 2 digits max after the decimal point?
Sometimes, the value exceeds the frame (???, 333333333333333333333333333 or ???, 66666666666666666666666666666667).
Thank you.
Otherwise everything else is fine, good job! 👍
[image]

gawindx, 2020-07-01:
this is corrected and updated in 1.8.0.1.
If my update function works correctly, all you have to do is update, otherwise, you just have to get the exe in the Build directory

Ririx02, 2020-07-01:
Perfect
Big thank

closed 2020-07-01 by Ririx02