gawindx/WinNUT-Client #20: Opening the settings dialog crashes the program

lucianm, 2020-06-30:
Hi, like mentioned in the subject, it's not possible to adjust anything in the settings window, the program then crashes after a second or so. Fortunately I found out (just seen this program today for the first time, as the UPS I ordered for my NAS / switch / tinkerboard with home automation arrived also today) that the program writes and also uses a configuration text file, where I could enter the host name of my NUT server. Starting WinNUT then works and the program keeps running, until trying to enter the settings again...

gawindx, 2020-07-01:
Rather surprising.

I see several possibilities:
- Corrupted exe, in this case try to download it again
- Problem in the Ini file, but as you tell me that you just downloaded it, and given that it is automatically created if absent at launch, I don't really believe it.
- A problem in the Dll file of the icons (the icons for the view / delete buttons of the log file are incorporated there). => No need to try to replace the dll manually, it is incorporated into the exe and automatically regenerated each time the exe is launched. So if it is corrupt, it is either the exe itself, or the exe on the repo.
- A problem with your system?

Do you have a particular error message? or does the app close without any explanation?

What you can try, and which would advance the problem, is to download [Autoit](https://www.autoitscript.com/cgi-bin/getfile.pl?autoit3/autoit-v3-setup.exe), the sources from the github and directly run the scipt WinNUT-client.au3.
It will not necessarily change the behavior of the app, but in the event of a problem not detected in the code, this should give more information on the file and the line that is causing the problem.

Another possibility, since you can edit the Ini file, is to manually activate the debug logs, start WinNUT and 'go to open preferences, then retrieve the file "Log.txt" in the same directory as the 'exe and provided via sharing or other ([Uploadedit](https://www.uploadedit.com/_to-upload-documents-onto-internet-PLAIN-TEXT-TXT-hosting.htm) for example)
It will allow you to know in which function it crashes.

To manually activate the log files, in the Ini file:
```
[Logging]
Use Log File = 1
Log Level = 8
```

USe Log File: 1 => activate / 0 => deactivate
Log Level: 1 => Notice / 2 => Warning / 4 => Error / 8 => Debug

Do not leave it active for too long in Debug as this will create a fairly large file over time.

To be honest, you are the first to report this crash to me, and I have no such concern on my machine (otherwise I would have fixed it before).

gawindx, 2020-07-01:
Another possibility that I hadn't thought of was a problem with a language file.
Try with a language other than the one selected, maybe it comes from there.
I tested on my side and no problem whatever the language but it may come from a translation that does not pass (a special character?)

lucianm, 2020-07-01:
Hi,

thank you for all the hints, I tried them all, enabling logging, changing all lagnuages, even installed Autoit and compiled the scripts again, for x64, also the icons DLL after removing the absolute path from the *.csproj, the behaviour was the same, and in the log there is no indication about the reason of crashing. Also, this scripting language unfortunately does not have a proper debugger, so I uninstalled Autoit again, for now.

gawindx, 2020-07-04:
sincerely sorry that the program has poor skills on your side.
and I don't understand where it can come from

since you manage to modify the ini file, I can only advise you to use this method.
if you need help modifying it, let me know, I will provide you with the necessary information.

just for information, I'm rewriting all of the software in Visual studio / Visual basic.
I think it should be less problematic afterwards

ThomasCr, 2020-07-14:
Hi, have downloaded the new release 1.8.0.2 - but there I got the same problem - when I open the settings - the app crashes.

gawindx, 2020-07-14:
To be very honest, since I do not understand the reason for this problem (I assume an AutoIt bug in relation to certain versions / Installations of Windows) and that version 2 is coming soon, I will not look into this issue.

Not that I don't want to fix it, but since version 2 will no longer be based on AutoIt, this problem should go away on its own.

I hope to be able to release v2 before the end of this week (at least in beta version with most likely the translations pending but which will be fully functional).

In the worst case, v2 should arrive before the end of the month with all the translations finalized.

But as it is a complete rewrite, I hope that I will not introduce regressions / bugs in the code.
In the meantime, the only way to work around the problem is to go through a manual edition of the ini file (it is automatically created with the default values on the first launch).

As a priority, the following keys must be modified:
```
[Connection]
Server address = Nut Server IP address
Port = Nut Server Port (3493 by default)
UPS name = UPS name
Delay = 5000
AutoReconnect = 1 (1 = automatic connection and reconnection in case of disconnection)
[Appearance]
Language = en-US (en-US / fr-FR / de-DE)
[Power]
Type Of Stop = 17 (17 => Shutdown / 33 => Sleep / 65 => Hibernate)
[Calibration]
Min Input Voltage = 210
Max Input Voltage = 270
Min Output Voltage = 210
Max Output Voltage = 270
Frequency Supply = 50
Min Input Frequency = 40
Max Input Frequency = 60
Min UPS Load = 0
Max UPS Load = 100
Min Batt Voltage = 0
Max Batt Voltage = 20
```
For the calibration part, the default values are suitable most of the time (in my opinion) but you can modify them according to your needs.

gawindx, 2020-07-16:
I just published the pre-release of v2.
There is still a bit of work, little things to refine on the setup, some functions still missing (but not blocking), translations remain to be done as well as the automatic import of an ini file (so as not to redo all the configuration.
The pre-release will therefore be in English.
At least this will allow you to have a functional product and, which I do not hope, will also allow to report possible bugs.

You can find this release here : [WinNUT 2 beta](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.0.0-dev)

gawindx, 2020-07-18:
The v2 had some problems, including the launch which did not necessarily go well.

I updated with several other fixes to v2.0.1.
I'll let you collect it [here](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.1.0-dev)

On the other hand, I won't be reachable next week so don't hesitate to give me a feedback if you discover things that are not going well.
I would modify them as soon as possible.

This version also includes translations for your language.
By cons there are some texts that are in the code that I must transform into resources to be able to make them localizable.

asitnik, 2020-07-29:
While I did not experience this issue with 1.8.x, version 2.0.2 crashed when opening the "UPS Variable" submenu.

gawindx, 2020-07-31:
Corrected

See here #23

gawindx, 2020-08-08:
I think that with V2, this problem is no longer relevant

I close the issue, it will have to be reopened if necessary

closed 2020-08-08 by gawindx