gawindx/WinNUT-Client #21: WinNUT only shows "UPS on battery" but it is actually on line

quadratmuede, 2020-07-11:
Hi I am using an Eaton Elipse Eco UPS on synology and trying to deploy WinNUT on some Windows PCs which is connected in the same network as Synology. After configuration, it shows the parameter properly, but the status (left panel) always says that the UPS is battery, but actually it is on line and 100% charged...

gawindx, 2020-07-11:
it is possible, but unlikely, that your inverter will use a different field or value than expected.
To check it, 2 possibilities:
- from the WinNut interface, file menu, UPS variable. This will display all the variables reported by nut for this inverter (no export for the moment)
- 2 method, from the server command line, the command upsc inverter_name, you will get the same result as for method 1 but in text format.

the expected value is "OL" for the ups.status field

if this is not the case for your inverter, I would be interested in you providing me with the text result in order to take this scenario into account.

on the other hand, if the value is good it will be necessary to see a little more in depth the logs in debug mode to see what value or error occurs on the processing of this data.

quadratmuede, 2020-07-11:
Hi there,

thanks for the quick reply. I checked it... Indeed Eaton uses different parameters... For status it uses "OL CHRG" instead of simply "OL" I will paste the wohle variables from upsc here:

battery.charge: 100
battery.charge.low: 20
battery.runtime: 2362
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 1200
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.version: DSM6-2-25364-191230
driver.version.data: MGE HID 1.33
driver.version.internal: 0.38
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 12
ups.mfr: EATON
ups.model: Ellipse ECO 1200
ups.power.nominal: 1200
ups.productid: ffff
ups.serial: 000000000
**ups.status: OL CHRG**
ups.timer.shutdown: 0
ups.timer.start: 0
ups.vendorid: 0463

gawindx, 2020-07-11:
okay
it can change without problem.

I could update it very quickly.

on the other hand I do not think I am wrong in saying that it is not a standard return.
it might be interesting to see the feedback you have in the different cases (on battery, partially discharged battery, or recharging) because I think "chrg" means a battery state.

jonathanrobichaud4, 2020-07-11:
Same here with me. I have "OL CHRG" as well. I have a CyberPower SX650U.

jonathanrobichaud4, 2020-07-11:
Once it reaches 100% charge it switches to on line.

gawindx, 2020-07-11:
thank you for this return, I suspected that it had a particular meaning

I will modify the code of v1.8 and take it into account for v2

gawindx, 2020-07-13:
The corrected version is available in [v1.8.0.2](https://github.com/gawindx/WinNUT-Client/releases/download/1.8.0.2/WinNut-client.exe)
It also includes a correction to v1.8.0.1 which had not been released.

closed 2020-07-13 by gawindx