gawindx/WinNUT-Client #22: Blocked by Antivirus Trojan

FrankyM, 2020-07-17:
Hi,
my GDATA Antivirus is blocking the winnut-client.exe -> Trojan.GenericKD. This happens with the original project and with your fork.
I checked the file with virustotal to see if GDATA is the only one - is not. Please see the attached screenshots.
Is there a possibility to fix this? I hope this is a false positive message.
Thanks!

ps: i can not upload the screenshots here, don't know why. Here is a link to my dropbox...
https://www.dropbox.com/sh/iqmg2tk4gl9lov0/AAB30yzdNvLQ2PW5ioZNPicJa?dl=0

gawindx, 2020-07-17:
The problem is known and being resolved.
Download this version, there is still a little work but it should be finalized within a week or two.
[WinNut_V2 Beta](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.0.0-dev)

closed 2020-07-17 by gawindx

gawindx, 2020-07-18:
I was told that the program was not launching correctly.
I updated the exe but it is necessary to manually replace the executable installed by the setup.

You just need to download the following file and replace winnut-client.exe in the installation directory (default: "C:\Program Files (x86)\WinNUT-client")

You can find the corrected executable here:
[Winnut exe functional](https://github.com/gawindx/WinNUT-Client/releases/download/v2.0.0.0-dev/WinNUT-client.exe)

sorry for the inconvenience

gawindx, 2020-07-18:
the beta version v2.0.1 is available [here](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.1.0-dev), it includes this morning's corrections plus a few others.

FrankyM, 2020-07-19:
Thank you very much. Yes it was not launching, now **it works!**

gawindx, 2020-07-19:
no problem

but it's still a beta version, there will surely still be things to refine

if you notice the slightest bug, don't hesitate to report it to me, I will correct it as soon as possible