gawindx/WinNUT-Client #23: Test 2.0.0.0 beta

Ririx02, 2020-07-17:
Hello,
I wanted to test the beta, so I put the ini file in the folder to transfer them. When I ran WinNut-client, I saw a message telling me that they have been imported, but the application does not launch. When I click on it I can see something launching in the task bar and in the notification bar but it stops immediately.

gawindx, 2020-07-17:
this is sometimes the problem in beta version ...

when you say "stop immediately", does the winnut process appear in the taskmgr or not at all?

I corrected a few points since because I noticed that randomly the program launches but that they refuse to be restored from systray.

if that's it, killing it and relaunching it was enough.

if the problem is more serious, go back to 1.8, I will be absent all next week and the project will not progress during this period.

gawindx, 2020-07-17:
if it is not indiscreet, I would like to see the contents of your ini file, just to see if it can come from there.
if I have time tomorrow, I will test and if so I will try to understand and reproduce your problem.

you can also try to delete the key in the registry and restart winnut then, the default settings will be applied.

The key in question:
HKEY_CURRENT_USER\SOFTWARE\WinNUT

Ririx02, 2020-07-18:
I tried on 2 different pc, with and without ini file and even deleting registry keys and reinstalling.
When I run it, I can see it in the task manager for a few seconds and then it disappears.
(Otherwise yes version 1.8.0.2 works perfectly but I still have this rounding problem in passing)
[ups.zip](https://github.com/gawindx/WinNUT-Client/files/4941120/ups.zip)

gawindx, 2020-07-18:
Indeed, after having tested the setup on a second pc, I encountered the same problem as you.

I added a functional version (in any case it starts correctly on this second pc) in the release, you just have to download it and replace the exe which is installed by the setup.
By default the installation directory is "C: \ Program Files (x86) \ WinNUT-client"

the direct link to get it:
[functional winnut exe](https://github.com/gawindx/WinNUT-Client/releases/download/v2.0.0.0-dev/WinNUT-client.exe)

gawindx, 2020-07-18:
in the problems that I noted, the log file is not created due to the rights of the installation directory.
I would correct this problem for the final version but I will run out of time to do so.

Ririx02, 2020-07-18:
Nickel, this patch will launch the application.
The graphs are much more stylish like that, well done.

On the other hand, I noticed something (not a problem when you know it) but the value of "UPS load" is not the right unit for me.
By doing the calculation of my side, I found that it is the value in VA. From memory, for a UPS, VA * 0.66 = W, and, when I apply this calculation to the value of the application, I obtain approximately the value indicated on the screen of the UPS.
On my APC Back-UPS RS 1500G, the returned value is 1444.

Small detail concerning the beta too, on "List UPS Variables", the "Reload", "clear" and "to clipboard" functions will crash the application.

gawindx, 2020-07-18:
Thanks for your return,

For the graphs, I also find it a little prettier and above all much more readable.

For the unit problem, you're completely right, I converted the value I had in W to the VA, instead of keeping it as it is, it's corrected.

For the "list ups variables" gui, I found where the problem is located and again I corrected it.

I added a release in v2.0.1.0.

I'll let you test it on your side, I won't be available next week so don't hesitate to give me your comments.

If it is stable, I will replace the dev branch with 2.0 and I will modify version 1.8 to take into account the paths to come for the update part.
I think I will let a little month go by to test the V2 well before making it final, it should leave time for 1.8 users who use the auto update to recover the 1.8 version with the correct download paths.

For v2.0.1, it's [here](https://github.com/gawindx/WinNUT-Client/releases/download/v2.0.1.0-dev/Setup.msi)

Ririx02, 2020-07-18:
2.0.1 tested, everything seems good to me now.
No more crashes and all values are correct.
Many thanks for your work!
I leave this topic open in case I find other bugs but obviously, everything is good.

gawindx, 2020-07-18:
It reassures me, I'm not too bad then :)

Ririx02, 2020-07-18:
You are even very good! 💪

asitnik, 2020-07-29:
Version 2.0.1 pointed updates available as version 1.8.0.3, and version 2.0.2 does not return any results whether positive or negative (e.g. "you're running the latest version", "no updates available" or another similar message) .

Ririx02, 2020-07-29:
same remark here, no message when there is no update?
I have crashes again when I go to the "UPS variables" option. As soon as I click on it, it doesn't even open the window for me, it crashes directly.

gawindx, 2020-07-31:
Problem corrected.
The 'UPS Variable' window no longer crashes the program and information on the availability or unavailability of a new version is correctly entered in the main window

That said, in the case of an available update, the window proposing the update is displayed, so the return to the main window is a little redundant.

The new version is available via the update (including for version 2.0.2, the verification is done correctly, it is just the feedback that was missing).

Ririx02, 2020-07-31:
everything is ok on my side 👌

gawindx, 2020-07-31:
I did not see any other problems either (that does not mean that there are not)

I will leave my development ideas in standby time to see if the version is stable before passing it on to the final.

I will still push the tests on the shutdown functions to be sure that there are no bugs left on this side.

thanks for your feedback.

jandari1, 2020-08-04:
Can you increase the battery voltage gauge to 30v? My UPS uses 2 12V batteries at 25V volts together, but only 18V max is shown.

gawindx, 2020-08-04:
normally, you should be able to set the values from the preferences in the calibration tab.

the values allowed for the battery voltage are between 0 and 999V, which should be more than sufficient.

that said, your question is not useless because by reviewing the values in the code, I realized that it must be possible to define inconsistent values such as minimum> maximum.

I would change that in the next version.

keep me informed if the modification goes wrong (there is no reason but in case ...)

jandari1, 2020-08-05:
OK I made this.
I have another suggestion. Qnap uses a username and password to connect to NUT server, when UPS hooked up to it. Can u add this feature?

gawindx, 2020-08-05:
@jandari1 
can you open a new issue for this feature?
it will be easier for me to follow

jandari1, 2020-08-08:
Maybe you can also add option to turn on balloon tooltip when ups switches to battery and back

gawindx, 2020-08-08:
it's doable, I hadn't thought about it but it seems relevant to me: function in the todo list

jandari1, 2020-08-08:
Oh and mybe you could make reconncting arrows in the Icon yellow. Would be easier to distinct.

gawindx, 2020-08-08:
I'm going to do the test to see if it would be more visible, anyway I had another icon to modify (that of the exe, as there is no outline, we cannot distinguish it on a white background )

casually, you start to create a nice todo list for me :)

jandari1, 2020-08-08:
Its a good program, why not make it widely usable and great! A lot of QNAP users are now using other Winnut, but your has great interface with a lot of options.

 Thanks for your effort.

jandari1, 2020-08-08:
Another suggestion: Add battery charge capacity on hover over icon in the status bar.

Ririx02, 2020-08-09:
In these very hot days, I have a few regular blackouts and I noticed that each time WinNut was shut down and I had to restart it each time.

gawindx, 2020-08-09:
@Ririx02 
Can you tell me a little more?
What do you mean by "blackouts"?
Do you have any particular information in the logs (if active)?

Describe me more precisely the course of the problem because it is a little vague as it is.

Ririx02, 2020-08-09:
When there is no more power, the UPS starts up and the app crashes at the same time.

ah and no I forgot to activate the logs >< sorry.

I activate them in debug mode and as soon as it starts again, I provide you with the logs.

Ririx02, 2020-08-11:
Hello,
Little feedback: I think it came from my pc, because nothing on the logs. I simply rebooted the pc (it had been running for quite a while) and I no longer had the problem.

jandari1, 2020-08-16:
I have tested the shutdown an power loss, but the system doesn't shutdown. When it should shutdown the WinNUT icon disappeared in tray, but nothing happened. Hera are my settings and log
[image]
[WinNUT-client.log](https://github.com/gawindx/WinNUT-Client/files/5080330/WinNUT-client.log)

gawindx, 2020-08-20:
hello everyone and thank you for your feedback.

the differences, problems or suggestions have all been taken into account and I am currently working on them.

For now, the new version is on track and should be available soon.
this version will also be the first stable version (hoping that I do not integrate new bugs ...).

Currently, I just have to integrate the identification and correct the problem of automatic extinction (as well as another small point of operation that does not suit me).

I'll keep you posted when the version is ready but it should be done before the middle of next week.

gawindx, 2020-08-27:
Version 2.0.4.0 has just been released.
It is officially the first stable release so I have to merge the dev branch with the master branch.

I think the majority of the problems encountered so far have been corrected or modified to get proper operation.

Therefore, I am closing this post because the beta test is now complete and any new problem will be the subject of its own topic.

Thank you for your feedback which allowed me to move forward in the right direction.

closed 2020-08-27 by gawindx