gawindx/WinNUT-Client #27: Username and password for connecting to NUT server on QNAP

jandari1, 2020-08-05:
I have another suggestion. Qnap uses a username and password to connect to NUT server, when UPS hooked up to it. Can u add this feature?

Some info here [https://www.reddit.com/r/qnap/comments/ehubuq/quick_network_ups_slave_setup_for_windows_using/](url)

gawindx, 2020-08-08:
I will need a little more information from you, because I am not sure where the problem is to need to connect.

From what I could read, the authentication will be required by Nut for the parameters of the inverter itself (delay / threshold of extinction for example).
WinNut is not planned (for the moment but I have already thought about it) to modify these parameters and, after the tests I was able to do on my server, even by defining a user, the consultation of the 'state (read only) is not blocked by this authentication.

In your case, do you have access to the status of your inverter (are the values ​​displayed in WinNut) or do you get a particular error?

When does Qnap authentication take place? on a web page? if this is the case, and you have access to the live state via WinNut, authentication is not required for me because we connect directly to the NUT server, but maybe I misunderstood your request .

Otherwise, in principle, this modification is perfectly possible if it is necessary and justified.

gawindx, 2020-08-08:
on proofreading, I wonder if there is not confusion between Winnut / Delphi for which you provided me the link and which seems more to be a pseudo port of the monitor part of NUT and this project, WinNUT-client which simply connects to the nut piur server to use the data, display it and control the shutdown of the client workstation if necessary.

jandari1, 2020-08-08:
As I researched Qnap uses NUT server for UPS, but its u/p protected. Not 100% sure. Maybe only thing you need to add authentication for connecting as you need to specify in the other Winnut. Im currently running NUT on my router and using QNAP as slave to connect to it and get to UPS data.
Only thing I can set in from the browser control panel is this.

[image]
[image]

gawindx, 2020-08-08:
ok, it's clearer like that

I will add this option and I will submit the exe to you for testing as soon as it is done (not sure how soon in the next few days but it's in the todo list)

gawindx, 2020-08-20:
thank you for your feedback.

the differences, problems or suggestions have all been taken into account and I am currently working on them.

For now, the new version is on track and should be available soon.
this version will also be the first stable version (hoping that I do not integrate new bugs ...).

Currently, I just have to integrate the identification and correct the problem of automatic extinction (as well as another small point of operation that does not suit me).

I'll keep you posted when the version is ready but it should be done before the middle of next week.

gawindx, 2020-08-27:
Version 2.0.4.0 has just been released.
I incorporated the possibility of identifying oneself with the Nut server.
This function remains experimental and is only active if a login and password are provided at the preferences level.

I'll let you test this and tell me if it works.

I relied on the official Nut documentation to implement it, I just hope that Qnap hardware authentication goes through this because otherwise it won't work.

jandari1, 2020-08-27:
Yeah, its working!!!

gawindx, 2020-09-07:
As it seems to work and you don't seem to have any problems on the connection, I think we can close the issue

closed 2020-09-07 by gawindx