gawindx/WinNUT-Client #31: Automatic startup

jandari1, 2020-08-28:
Now im having a bug that the application does not start up even enabled start up with windows. Its also visible in windows task managers start and its enabled. When I copy the link to start menus run folder it starts up normally. I also reinstalled and deleted all registry entries and folders before and no change.

gawindx, 2020-09-07:
I didn't have time to answer you but did you solve your problem?

For the notification suggestion, I note for a later update (in fact as soon as I have enough time to manage it)

jandari1, 2020-09-07:
No. I copied link to startup folder in start menu, then it starts normally.

gawindx, 2020-09-08:
To manage the automatic start, WinNUT creates an entry in the registry at the current user level.

is it possible that you have run WinNUT as admin or from another login?

look in the registry if you can find the key in question here:

HKEY_CURRENT_USER\SOFTWARE\Microsoft\Windows\CurrentVersion\Run\

optionally, you can try to delete it manually (only the WinNUT key) and see if it works again correctly once recreated from the WinNUT preferences.

quick question in passing, does it have to do with a change of version of WinNUT? (I have not had any feedback in this direction but I prefer to rule out this hypothesis)

jandari1, 2020-09-08:
I have checked. The key is added to registry, but it does not start up. My account is the only one on computer.

gawindx, 2021-02-15:
A little up to know if you still encounter this problem or if we can close the issue.

jandari1, 2021-02-20:
Seems to be fixed in new version it works as it should.

gawindx, 2021-02-20:
ok, having never been able to reproduce a problem I could do nothing to solve it.

But the important thing is that it works.

I close the issue

closed 2021-02-20 by gawindx