gawindx/WinNUT-Client #32: Not responding to force shutdown FSD test command

FileCity, 2020-08-31:
The application is not responding to the upsmon FSD command sent from the server (upsmon -c fsd ).
According to the log, the command is received but WinNUT just sits and waits to get disconnected.
WinNUT should initiate the forced shutdown of the host when this command is received from the NUT server.
Please find a log attached, search for: Process Result With ups.status : VAR UPS ups.status "FSD OL"
Thanks to have a look at it and I'm available if you need to test something.
[WinNUT-client_FSD_Fail.log](https://github.com/gawindx/WinNUT-Client/files/5147753/WinNUT-client_FSD_Fail.log)

gawindx, 2020-09-03:
indeed, this information is not processed (and to be honest, I had not yet studied the NUT protocol documentation so thoroughly)

The change will be made on a next update, but we will have to wait a bit because I am sorely lacking in time at the moment and this change will be made at the same time as the evolution for the automatic configuration (roughly, except value no managed by the inverter driver, it will no longer be necessary to manually calibrate the min / max).
This evolution obliging to radically modify the class managing the communication with Nut, this will be the right moment to integrate this functionality (taking into account that it will be necessary at this moment to implement all the gaps in the NUT protocol which dates from versions 1.x of this project. )

FileCity, 2020-09-03:
Thanks a lot for the answer. The FSD (force shut down) flag is very important in server/workstation environment because it is the most effective mechanism for NUT to tells all client to shutdown immediately in case of a significant event.
For now, I will configure the WinNut machines to shutdown at a higher battery level to avoid issues. Other Linux machines and my pfSense firewall can manage the FSD well.
Another thing that can also be integrated is the 'in alert' pool delay. Normally we pool at 5-10 seconds but when the UPS is on battery, most system admins prefer to pool at each second, to not miss the FSD flag in case of an issue or failure.
NUT can manage these two pools delays well at client side.

Another major improvement that can be implemented is to have the core section of WinNut to run in a system service and the GUI that fetch the info from that service.
Here is the main advantage: when a server shuts down after draining the battery, when the electricity returns and the UPS have sufficient battery level, the BIOS on a server is sometimes configured to power back ON the machine automatically.
This way, the system returns to a working condition automatically, but WinNut may not run if a user is not logged-on at that time... maybe...
The system service will run and ensure the core/system part of NUT is working and doing it's job.

I have also noted a date issue in the log file, at each new section the date is always the same, must be the current one and incrementing to follow log entries.

And at last, when WinNut tried to reconnect to the NUT server the GUI is not always responsive, freezing.

If you need help or want to test something, I've access to many UPSs, Virtual Machines, etc...
Do not hesitate to ask. Thanks for you help, WinNut is now really promising.
Keep the good work.

gawindx, 2020-09-07:
I come back to you as soon as I move forward for this part of the development.

The possibility of having access to several UPS would greatly help me for my tests, but this is unfortunately not the case for me.

For the system service, it would not be a bad thing but I am not an expert in programming either, I am passionate; this development will therefore risk being quite long for me (but that does not mean that I will not do it because this idea fits quite well with my idea of managing some graphs allowing to see the evolution of values over time and that would not necessarily have the GUI part loaded. The service could then take care of collecting and storing the data)

For the freeze when reconnecting, I also noticed it.
I think I would have to manage this in a second Thread to avoid the problem, but again, it's a notion that I don't understand well.

FileCity, 2020-09-07:
Merci de votre réponse, et oui je suis Québécois d'origine francophone :)
If you need something, just ask.

gawindx, 2020-09-07:
C'est bon à savoir, étant français, je pense qu'on se comprendra mieux dans la langue de Moliére.

Mais pas de souci, dés que j'ai du temps pour avancer sur le projet et que j'ai une ébauche qui tient la route, je reviens vers toi.

FileCity, 2020-09-07:
Merci. Et comme je dis souvent, la meilleure façon de manger un éléphant est une bouchée à la fois.
Fait-moi signe si tu as besoin.
Je suis technologue en électronique avec une sérieuse expérience en informatique, je ferai de mon mieux pour aider si cela est nécessaire. Je code un peu mais surtout en assembleur et AutoIT pour automatiser des tâches, on va oublier ce dernier :)
Bonne fin de journée

gawindx, 2020-09-07:
Je confirme, pour AutoIT on ne vas pas trop s’étendre sur le sujet.
Personnellement, je n’ai rien contre ce langage mais je pense que, compte tenu de sa mauvaise réputation vis à vis des antivirus actuel, il n’est plus adapté à un développement distribuable et il vaut mieux le réserver à une utilisation personnelle ou « confidentielle » ; raison pour laquelle j’ai tout réécris en VB.
Pour ma part, j’ai aussi de bonne base en électronique mais côté informatique j’y suis revenu sur le tard (j’ai commencé avec le Basic donc ça date un peu, voir beaucoup).

Par contre, j’ai beaucoup aimé ton expression, je l’avais jamais entendu.

Côté développement, je viens de changer de boulot, raison principale de mon manque de temps actuel et je travaille aussi sur un autre projet sous Homebridge, j’essaie donc alterner d’un projet à l’autre pour que les  deux puisse avancer mais ça n’aide pas non plus.

Je reviendrai vers toi mais je pense qu’il faudra attendre quelques semaines avant une réelle avancée.

Je te souhaite également une bonne journée.

gawindx, 2021-02-09:
La command e  FSD est maintenant supportée.
Pour le délai de rafraichissement, je n'ai pas fait de modifications pour le moment, je m'interroge encore sur la pertinence de cette modification.

Je ne ferme donc pas l'issue en attendant d'avoir tranché sur ce point précis.
LA nouvelle version sera dispo sous peu.

FileCity, 2021-02-21:
Est-ce que la commande FSD demande un test supplémentaire ?
Je pourrais tester avec un 2e serveur NUT pour valider, cela reste un composant critique qui doit fonctionner si il est intégré.

gawindx, 2021-02-21:
Du moment que l'option est activé dans les préférences, cela devrait fonctionner.
Mais si tu souhaite le tester, pas de souci.

Au contraire, si ça ne marche pas je préférerais le savoir mais je l'ai forcée de mon côté depuis le serveur NUT et ça réagissais correctement maintenant, on est pas à l'abri d'une particularité.

Et je suis en train d'intégrer le rafraichissement plus rapide sur batterie.
Ce qui m'a permis de voir que cette partie du code n'était pas très adaptée pour gérer plusieurs informations différentes de la variable ups.status et qu'au fur et à mesure que des informations supplémentaires doivent en être extraites, ça devient vite le foutoir.... C'est modifiée.

Je publie aussi une nouvelle version corrective suite à un bug sur les mises à jour qui m'est remonté tout à l'heure.

FileCity, 2021-02-21:
@gawindx, je viens de tester le FSD avec un 2e serveur NUT et une VM, le shutdown à été immédiat.
Looks good :)

gawindx, 2021-02-21:
Ok, et comme je viens d'intégrer l'accélération du temps de rafraichissement sur batterie, je pense qu'on peux considérer que cette issue est terminée.

closed 2021-02-21 by gawindx