gawindx/WinNUT-Client #33: NUT Server doesn't see client, but client sees NUT UPS Data

aaronmweiss, 2020-09-05:
Just installed this. Awesome so far.

I am able to get the WinNUT-Client see the UPS data, however I am unable get the NUT Server to see the Windows machine via ```psc ups```.

I want to confirm that the WinNUT-Client is able to proceed with shutdown parameters even if the Server doesn't see the client.

FileCity, 2020-09-05:
Hi aaronmweiss, actually WinNut doesn't implement all of the NUT protocol.
I will let gawindx to explain and discuss this part.
WinNut will fetch the UPS data from the server only and actually not responding to commands sent from the server.
A workaround is to set the shutdown battery level higher to let the Windows machine shutdown by itself before other Unix/Linux/BSD clients as those will react to server commands...
Clear enough ?
Let me know if you need more assistance.

aaronmweiss, 2020-09-06:
Thank you. Super clear. I took your advice and increase the shutdown threshold higher than my other machines. 
Still would love to hear more from gawindx if he as more information to provide.

closed 2020-09-07 by aaronmweiss

gawindx, 2020-09-07:
I couldn't have said better.

The FileCity solution, in the absence of better protocol support, seems to me to be the best solution for the moment.

This evolution of the project is still young and I started from (almost) zero because the old project suffered from AutoIt's bad reputation.

There is still a lot of work to be done and I think that I have not yet put "my hands in the grease" enough to fully understand all the workings of the NUT protocol.

But the internet is my friend and I have already found a lot of information which should help me to develop the project in the right direction.