gawindx/WinNUT-Client #34: [Feedback/Info for Freenas users]

DemonfangArun, 2020-09-07:
Since I ran into this I thought I'd leave it here for others:

Set the Delay to 1, then it will actually connect, 0 doesn't work. Poking settings helps :D 

It might be worth putting this into the wiki

gawindx, 2020-09-07:
thank you for this feedback.

my english is far from perfect and i'm not sure i understand your words.

concerning the delay of 1 or 0, except for the refreshment delay, I am not sure to see what delay you are talking about.
if it is the one, it is in milli seconds and its default value is 5000ms (5 seconds).
I did not manage to reproduce it but I suspect bad behavior of WinNut which causes a resetting of all the values ​​instead of resetting them by default (or of doing nothing at all for that matter).
Maybe you have encountered this problem.

Regarding the shutdown delay, there are normally 2 values ​​that are concerned.
the first is a minimum% of battery below which the stop sequence is initiated, the second value is a minimum hold time below which the sequence is also initiated.
The first of the 2 conditions reached (subject to being on battery) initiates the sequence and the 2 values ​​must be above their respective threshold to automatically stop the sequence (or a return to mains voltage).

It's true that I haven't done a wiki yet, that will be one of the upcoming points on which I will have to work.

DemonfangArun, 2020-09-07:
The refresh delay, from the options. the default value of 0 results in what I'm guessing is it trying to get the info before it's fully connected, and it fails and doesn't do anything. I'm no programmer but it's not the first time I've run into this sort of thing happening.

Regarding the delay, I could've been more clear. An option to choose between minimum battery % or start shutting down after a specified number of seconds on battery. A way to use both might not be bad (such as if power came back but battery was low then power was lost it would immedately shutdown, overriding the timer). This is unrelated to the "Delay to Shutdown" setting.

gawindx, 2020-09-07:
Ok, I understand better what you mean.
For example after 5 min on battery it is considered that it is better to start the shutdown procedure to avoid over-discharging the batteries unnecessarily.
Could the goal be to leave as much energy as possible to allow the more greedy server time to stop or to continue to function as long as possible (considering that the Windows workstation or WinNut is running is not vital)?

For the refresh time, I can confirm that a value of 0 is absolutely not normal and that a value of 1 forces WinNUT to refresh the data of the inverter every milliseconds (but this work).

Clearly this is not a good value - I should change the min value allowed to 100ms (which would be more than enough).

The default is 5000ms and I would advise you to set a value closer to the default (you can set a value between 1 and 60000ms)

DemonfangArun, 2020-09-07:
It's two fold. The first is that this UPS also runs networking equipment which during a power outage is more important to me than the servers being up. The second is that applications get very much confused when shares they are expecting to be there suddenly aren't, leaving me to have to manually intervene and fix things. Sequencing is important.

Thanks for clarifying what unit that delay is in, I'll change it to something reasonable, 5 seconds is plenty fast enough.