gawindx/WinNUT-Client #35: [Translation] zh-CN Simplified Chinese

nonPointer, 2020-09-15:
[Gist Link](https://gist.github.com/nonPointer/813d54c81f4582b9ac0d532d23e3dd69)

gawindx, 2020-09-15:
thank you for this translation.
I integrate it into the next version and then I will close the issue.

just to be sure, this translation only concerns the zh-CN locale and is not compatible with the zh-TW locale, is that correct?

nonPointer, 2020-09-15:
Thanks for your response.
Yes, this translation only concerns the zh-CN locale in Simplified Chinese, while a zh-TW translation should based on Traditional Chinese.

gawindx, 2020-09-15:
ok, that's what I understood and as I don't know Chinese, I preferred to be sure rather than badly integrate the translation

nonPointer, 2020-09-15:
Thanks for your careful consideration.
Since the grammar various among different languages, some sentences may sound weird in my translation. It could be better if you could add more {} tuple for variables. For example, "Shutdown if battery lower than"=>"Shutdown if battery lower than {0}" so that it can be translated to "在电池电量百分比低于 {0}% 时关机". but not "如果电池低于百分比则关机"(the grammar is weird in Chinese). Furthermore, I would like to recommend you put a combined CSV file under the repo so that others can simply refine the translation by pull request, instead of creating a full translation document.

gawindx, 2020-09-16:
For the csv, this is not a bad idea, it would indeed be easier to manage.
It remains to be seen how I can use it more easily with Visual Studio.
Otherwise I will have to manually copy each translation (which I am already doing).

For the tupple, I'll see what I can do, I can't guarantee anything but I'll see.

gawindx, 2021-01-18:
I followed your idea and created a csv file to make it easier to update languages.

I also had to make some modifications / additions to the texts.
Which implies that your translation may not be perfectly exact anymore.
Can you take a look and correct if necessary - I will release version 2.0.5 soon but I am waiting for your feedback to do so

nonPointer, 2021-01-20:
Thanks. I would check it asap.

nonPointer, 2021-01-23:
I think the "Devellopement" row in the CSV is a typo.

gawindx, 2021-01-23:
oops
well seen
English is not my native language, and you can see it ....

gawindx, 2021-02-09:
I had to edit or add some text.
I also modified the csv file to have a better follow-up of the translation status (you will see a status field as well as a field indicating who made the translation).
This way, and if you want to update or correct some translations, it will be easier for you and me to follow.

gawindx, 2021-02-13:
Pull requests validated, but I continued the development and raised a lack on a function so, if you can translate me 2 additional sentences (to avoid a new pull request) :
1 - Windows standby. WinNUT will disconnect.
2 - Restarting WinNUT after waking up from Windows.

nonPointer, 2021-02-16:
Here is the translation.
1 - Windows 待机。WinNUT 将断开连接。
2 - 重启计算机后自动运行 WinNUT。