gawindx/WinNUT-Client #37: NUT client password in plain text and visible

FileCity, 2020-09-21:
In the configuration, the password of the nut account is visible in plain text.
This can be an issue on shared accounts. Passwords are always sensible...
The password should be entered and then replaced with dots when viewing the configuration.
Should also be encrypted in some way in the registry if possible:
Computer\HKEY_CURRENT_USER\Software\WinNUT\Connexion
NutPassword key is in plain text.

Thanks.

gawindx, 2020-09-21:
I both agree and disagree with you (which doesn't mean that I don't want to make this type of data more secure).

That the password is in clear in the registry is indeed a risk.
That the password is clear in the application is not really one in the sense or, even in a multi-user environment, because if each user has his own session, he can only obtain the password by the registry is then a matter of Windows rights or of not displaying the password when a third party can look over the shoulder.

But it is clear that it should not remain as such and that I must improve this point.

The problem is that this will only be an ofuscation because I need to be able to read the password again to send it to the NUT server.
It will therefore not be readable directly but, by using the source code, it will be possible to recover it anyway (and my project will remain free so access to the source code will remain possible).

I will still study this point and see if a non-reversible method (by a human) is possible.

gawindx, 2021-02-09:
Features present in the new version to be released today.

I close the issue.

closed 2021-02-09 by gawindx