gawindx/WinNUT-Client #39: Ability To Run Scripts When UPS State Is Changing

FileCity, 2020-09-28:
It should be a good idea, on a future release, to add the ability to run scripts when the UPS state is changing.
On battery = Script1.script
Back Online = Script2.script
With this we would be able to stop power hungry processes to save the batteries and resume them when the utility power returns. This is possible on Linux and it's part on the NUT package.
This can be added from inside WinNUT-Client application or triggered from a system event with the Task Scheduler ( this approach is more easier to integrate at first). Actually WinNUT-client is not registering any system events when the UPS state is changing so it's not possible for now.

Thanks.

gawindx, 2020-09-29:
it was already planned in one side of my head, but I think it won't be for now.
I consider it a comfort (at least for now) and other modification seems more urgent to me.

In addition, adding this feature would not necessarily be in line with the changes that await me and could complicate my task.

But yes, it was planned.

FileCity, 2020-09-29:
@gawindx, no problem. It can be a nice feature to stop some services, like GPU computations, while on battery.
I fully understand your point and it's always better to add and debug features incrementally than all at once...

sw90036, 2023-06-11:
I am running the NUT Server on a Synology NAS so I don't have any options to run processes from the server.  I have a set of PowerShell scripts (which I will be happy to contribute) that I would like to run at various points while the UPS is on battery so I am looking forward to seeing this!  THANK YOU for your efforts on this!

gbakeman, 2023-06-11:
Everyone,

This repository is abandoned and no longer developed. I've reopened this issue at the current repository, [here](https://github.com/nutdotnet/WinNUT-Client/issues/92) . Please close this current issue if possible and subscribe/participate in the one I linked.