gawindx/WinNUT-Client #40: winnut crashes on close

chrcoluk, 2020-10-02:
version 2.0.4.0

unknown software exception

gawindx, 2020-10-02:
To be honest, I would need a little more information to try to understand what that could do to WinNUT.

Now, I understand that the error you encountered is not very meaningful either.

You opened another issue concerning a connection problem, we will already try to solve this connection problem before looking into this error problem when closing.

Subsidiary question: does this error happen every time you close (in which case we will know if there is a link with the connection problem if we solve it and you no longer encounter the error when closing) or is it an isolated event (and in this case it will be a little more complicated to define)?

chrcoluk, 2020-10-02:
If I close via exit, there is no crash, only using X icon.

gawindx, 2020-10-02:
For the other projects, I don't know but even if I don't have a lot of time to devote to it, I will not abandon it before having a correctly completed tool (and it is not for as much that I will give up elsewhere).

I will see if I see why you are encountering this error but as far as I can remember I have not encountered this problem and no one has reported it either.

A more advanced version correcting many small "bugs" or inconsistencies is under development, there is a good chance that the problem you encounter will go away on its own.
In the meantime, I cannot advise you too much than to avoid using the cross to avoid the error (I know, it's a patch, but I have nothing better to offer you for the moment).

Can you tell me which options are checked in the options for this tab:

[image]
Because the behavior of the closure changes depending on these options

chrcoluk, 2020-10-03:
I have everything ticked except "Close to tray", the 3rd option.
Logging level is currently Notice.

gawindx, 2021-02-15:
Due to lack of time to work on it, the new version was delayed.

It has been available for a few days and maybe you already got it.

If you can give me a feedback to see if you still encounter this problem with the new version.

closed 2021-03-15 by gawindx