gawindx/WinNUT-Client #41: winnut does not connect to master

chrcoluk, 2020-10-02:
No firewall rules blocking client or server side

I have enabled log in debug mode and there is nothing there indicating why it is not connected.

Just this

02/10/2020 16:47:02 11188  WinNUT : Force Reconnect from menu
02/10/2020 16:47:02 11188  UPS_Network : Connect To Nut Server

I do use a username and password which is experimental?

chrcoluk, 2020-10-02:
Ok this is fixed by changing the delay to 5000 from the set 0.

closed 2020-10-02 by chrcoluk

gawindx, 2020-10-02:
Okay, I see you anticipated my answer.
This problem has already been encountered (concerning the value of 5000 instead of 0) but I have not yet figured out why the values ​​are found at 0.

To answer your other questions and for the record concerning the test methodology for a possible network problem (information that I will have to put in a wiki when I have created it):

The use of a login and password is not an obligation.
It all depends on whether the user defined in the configuration of WinNut requires authentication or not.
This option has been implemented to allow users with a Qnap NAS to be able to connect directly to the on-board NUT server (and it is likely that other users will also need this option).

For my setup, this option was never required and I think most simple setups don't require it.

Regarding the connection problem itself, WinNUT makes a "simple" connection directly to the port / IP of the Nut server.
An easy way to find out if the problem is with WinNUT or a network / NUT server problem is to test a telnet connection on the Nut server.
If the connection is successful, the problem is with WinNUT.
If it is not possible to telnet to the Nut server then the problem is not with WinNUT.

Basically you open a cmd window, you run
`telnet IP_Server_Nut Port_Server_Nut`

If you get a black screen with just the cursor try the command
`LOGOUT`
The nut server should answer you "OK Goodbye" (in any case, it works like that with mine)
You can also try other command such as:
`LIST VAR your_UPS_name`

This should allow you to retrieve the list of variables managed for your UPS.

And finish by `LOGOUT` for a clean disconnection.