gawindx/WinNUT-Client #42: Request: Ability to run as a service

oyvsi, 2020-10-20:
In order to use this on servers I need to be able to start this automatically and not in the context of an interactive user. I guess this would also mean splitting the GUI-parts from the application.

gawindx, 2020-10-20:
This will be done, the development of this feature is already underway, the request having already been made and it seems to me to be quite relevant.
On the other hand, I have at this moment no idea of the delay of development.
We will therefore have to wait until then.

polcape, 2021-02-21:
+1 for me

gawindx, 2021-02-21:
This functionality will come but in the meantime I have reviewed my roadmap and it is not for now, there is still a little work before I can get there.
The next big step of the project is to deport the common functions/classes to the GUI and to the service to a DLL so that it is available to each part.
Only then, and once resolved any problems that the previous step may cause, I could move on to the development of a service that will manage the communication between the gui / Windows and NUT (or only NUT and windows if the gui is not not executed)

manjul968, 2021-04-07:
+1 much needed function!

Sp0ngee, 2021-08-08:
+1 so need this function please!

myriaxis, 2022-07-18:
+1 for me as well.
In the mean time, is there anther option?

rayrayrayraydog, 2022-07-28:
I think the best option would be the complete port of NUT server to Windows that is in Beta here: https://networkupstools.org/download.html

myriaxis, 2022-07-28:
There is not much documentation that comes with it.
I finally decided to use AlwaysUp to run WinNUT as a service. The downside is that the UI icon is not present in the notification area, but if I need the interface I can always manually start it while I'm logged in.
A free version would be to use Windows Task Manager to start WinNUT at Windows start.

sphinxicus, 2022-08-19:
+1 for this functionality - I have a windows host running a VMS which starts as a service, it's protected by a UPS but i very rarely need to log into the machine.