gawindx/WinNUT-Client #43: WinNUT variables reporting incorrectly

MattHorn25, 2020-10-26:
Hello! I've got WinNUT set up to connect to a Synology DS918+ NAS which is attached to a CyberPower CP1500EPFCLCD UPS.

The NAS seems to be reading the UPS state correctly, and WinNUT _used_ to as well, but it's recently started reporting the battery charge at 20% (the UPS is always online and at 100% and reading from the NAS shows this) and the UPS load at 100% (it fluctuates at around 10-30%). The UPS details show as Unknown, and the 'List UPS Variables' window is blank, even when I click 'Reload'. Disconnecting from the NAS and reconnecting does nothing.
[image]

Any idea what might be causing this? Thanks in advance!

FileCity, 2020-10-26:
I've heard of many issues on the past with UPSes and NAS units.
Questions, where is the UPS plugged-in ? USB or network management card ?
Where is the NUT server, on the NAS or installed on another NUX machine ?
According to your screenshot, WinNUT is not correctly fetching the info from the UPS...
The NUT server itself can also be incorrectly configured or not compatible with your UPS, unsure...
Please provide more information about your specific setup.
Thanks

MattHorn25, 2020-10-26:
The UPS is plugged in over USB to the NAS, this is a small office/home office UPS model that I don't think has a network management feature at all. The NAS OS shows the correct device information and status. The UPS has its own display so I know its status matches what the NAS is showing, and doesn't match what WinNUT is showing.

The NUT server is on that same NAS, Synology NASes have it built into the OS - it's a regular NUT service, but it's usually meant to be connected to by other Synology devices. This means that there are no configuration options, all Synology NASes run the same NUT server with an identical setup. You can only turn it on or off. Let me know if you have any other questions, hope I can help!

gawindx, 2020-10-27:
the problem you are having could be caused by an incorrect refresh value.
in preferences, check the value "refresh" (under ups name un first tab)
this value is in ms and must be 5000(5sec) by default.
there is a bug in winnut which causes bad initialization of values during the first launch and it happens that the value is 0.

MattHorn25, 2020-10-29:
Just for future reference, the "Delay" value in Preferences was already correctly set to 5000.

I've managed to resolve the issue! And it seems to be mostly a Synology problem, not a WinNUT problem. Unfortunately I didn't do it step by step so there are two things that may have fixed it. If this happens to anyone else, try these two things:

1) Disable the 'UPS support' entirely on the Synology NAS (uncheck the "Enable UPS support" box in Hardware & Power settings, click Apply) then enable it again (check the box, click Apply).
2) Make sure to click the "Permitted DiskStation Devices" button on the same page, and confirm the IP of every device/computer with a NUT client that's connecting to this NAS is entered there correctly. Disabling then enabling UPS support will clear the list.

I've reproduced the error:
When the UPS service on the Synology NAS is turned off, or the NUT client IP is not listed as Permitted, the NUT server on the NAS _seems to still respond and the NUT port is open - it's just not reporting values properly,_ but WinNUT displays default values anyway as if it's working. I'd ask that if WinNUT encounters completely bogus values like this it throws an error instead of failing silently.

I'm attaching a debug log file to show what the connection looks like when trying to connect to a Synology NAS NUT that isn't responding.
[WinNUT-client.log](https://github.com/gawindx/WinNUT-Client/files/5456942/WinNUT-client.log)

Thanks again!

gawindx, 2020-10-29:
just a hypothesis, wouldn't it be an identification problem that causes the data access problem?

otherwise, I confirm that, as winnut cannot recover the data, it applies the default values ("Apply fallback value" in your logs), and this in order to make it possible to calculate the missing data from autonomy.
where it gets stuck is that as it does not recover any, it applies them all.
I will have to improve this to limit to one or two max values in order to avoid this scenario.

MattHorn25, 2020-11-03:
If by identification you mean a whitelisting problem - yes, apparently Synology NUT servers do need to have the IP of every client whitelisted, so I believe the issue was caused because a new unlisted IP was assigned to the PC via DHCP after a router reboot.

Thank you for the followup.

gawindx, 2021-02-09:
I think the problem you encountered is no longer relevant, so I close the issue.

If I'm wrong, don't hesitate to reopen it.

closed 2021-02-09 by gawindx