gawindx/WinNUT-Client #45: Wrong information in the tray icon when plugged in

Alian51, 2020-11-08:
With mouse over the tray icon, the mention On Battery (xx%) is mentioned even if the UPS is on AC (at least in the French version, maybe only a translation mismatch).
The tray icon is well reacting when plugging / unplugging the UPS on / from AC, becoming green or red, but the indication still showing On Battery.
BTW, very good app!
[Tray icon mouse over.docx](https://github.com/gawindx/WinNUT-Client/files/5507228/Tray.icon.mouse.over.docx)

gawindx, 2020-11-09:
this is not a bug in the sense that, even if your ups is plugged into the mains, your battery is not necessarily 100% charged (which is the case on your capture), especially after a power cut .

In addition, the fact of displaying the state of charge also makes it possible to see a battery problem coming, such as a battery which no longer takes the charge correctly or which does not maintain its nominal voltage (and therefore its charge).

gawindx, 2020-11-09:
I read a little quickly (and badly because, like you, I am French).
indeed, this is not normal.
that the battery level is always displayed is normal, but that we are all the time on battery even connected to the mains is not.

has to be corrected for the next version (and it is indeed possible that this is a translation error).

Alian51, 2020-11-09:
Thanks for your feedback ! And looking forward to installing the new version as soon as released!

gawindx, 2021-01-12:
I come back to you concerning this problem (I am putting my hands in the code so I am taking advantage of it).

It seems that your problem is not due to a translation problem but rather to a bad treatment (of my fault therefore) of the return value of the state of your ups, value which seems to be returned with an unnecessary space (thus fault of your inverter or NUT server).

What is stupid is that I had bypassed the problem for the value of the icon but I did not think of doing it for the text ...

this will be fixed in v2.0.5 (should be released soon).

Alian51, 2021-01-13:
Great!

Thanks for the follow-up.

 

Enjoy your week.

gawindx, 2021-02-09:
The functionality is present in the new version, I close the issue but do not hesitate to reopen it if it does not correspond exactly or it can be improved.

closed 2021-02-09 by gawindx

Alian51, 2021-02-12:
Great! Just tested and it works!
Thanks for the folow-up.