gawindx/WinNUT-Client #47: How to setup WinNUT Client on Win 10 with synology

witzker, 2020-12-10:
I put in the NUT Host IP
As port for Synology 3493
Smart-UPS C 1500 as shown in Synology Host

username and PW for Synology Admin

When I click connect nothing happens.
Pls Help

tgp1994, 2020-12-23:
When I was setting up the original NUT software to talk to a Synology host, I found out that Synology uses a hardcoded username/password to connect to their NUT software. In /etc/syno/etc/ups/ups.users, the username is _monuser_ and password is _secret_. Although when I just tried this with WinNUT, it did not indicate a successful or failed connection, so I think it may be silently failing somewhere.

FileCity, 2020-12-23:
I always setup my NUT server (with the UPS connected to it) on a Linux machine and then I setup each clients.
You can use a Raspberry Pi as server, works very well and it's using the complete NUT package with all options available.

tgp1994, 2021-01-10:
Update for @witzker: I was able to get this working with my Synology device.

On your Diskstation web page, go to Control Panel, Hardware & Power, then click on the UPS tab. Make sure **Enable UPS server** is checked, then click the **Permitted DiskStation devices** button. Type in the IP address for your computer running WinNUT. You might need to put in both IPv4 and IPv6 addresses, I'm not clear on which one is preferred. After you've done that, click OK and then Apply. Now open the Options for WinNUT. Type in the hostname or IP for your diskstation in **NUT host**. Port can be left at 3493, and UPS Name left at UPSName. For **login**, type `upsmon`, and for **password**, type `secret`. Click OK, and it should connect.

Please let us know if you have any issues with this.

gawindx, 2021-01-12:
@tgp1994 
Can you take some screenshots and post them? I will have to do a wiki one day or another and that will allow me to inform specific cases.

tgp1994, 2021-01-13:
Sure! Here's the first part:

[image]

1. Navigate to DiskStation's webui
2. Open Control Panel
3. Find the Hardware & Power section
4. Open the UPS tab
5. Make sure Enable network UPS server is checked, then click Permitted DistStation Devices
6. Put in the WinNUT's client IP address, and click OK
7. Finally click Apply (important!)

Finaly step, in WinNUT:

[image]

1. Open WinNUT, and open the Settings menu.
2. Put in the hostname or IP address of your DiskStation.
3. Set login and password as I've set
4. Click OK, and it should connect.

SeeBer61, 2021-01-17:
I've just recently set up my UPS (APC Back-UPS Pro) to secure my Diskstations. I've set up UPS monitoring with automated shutdown (to safe mode) in my DS220+ as a server and DS213+ as a client. Now I was looking for a way to include my PCs into this and found this project. Great !!
Actually it took my only one hour to set everything up including configuring Windows NUT client. After some trial & error I did some research and found that only
**ups**  works as a parameter for the _UPS Name_
**upsmon** as _Login_ and **secret** as _password_ is still the default config in DSM6.x

tgp1994, 2021-01-17:
You're right - I don't know why I didn't figure it out before, but I had the ups name wrong the whole time until recently. I'll need to redo the screenshots...

gfairchild, 2021-02-23:
I'm trying to get WinNUT 2.0.7721.35955 working with my Synology DS1019+, and I'm having issues getting WinNUT to talk to the NAS. I'm hoping some of you guys might be able to help.

As you can see here, I've got the UPS attached to the NAS:

[image]

I've got my PC's IP included in the list of permitted devices:

[image]

I've verified my PC's local IP both in my router's device list table, as well as in my Windows networking settings.

I've plugged everything in WinNUT as best I can:

[image]

Here, I've tried both `monuser` and `upsmon` as the login, and `secret` is the password. Additionally, I've verified the IP of my NAS, and I've also tried the NAS' hostname.

When I try to list UPS variables, I get this error:

[image]

`[Id 7: 2/22/2021 9:26:54 PM] Connection to Nut Host 192.168.1.46:3493 Failed: Connection to Nut Host seem broken when Retrieving ListUPSVars`

Does anyone have any ideas? The firewall on the NAS is completely disabled, so it shouldn't be that. Any help is appreciated!

gawindx, 2021-02-23:
Try With « Polling intervalle » >=1 in your options.

gfairchild, 2021-02-23:
> 
> 
> Try With « Polling interval » >=1 in your options.

Ok, that's hilarious; that worked immediately! I should've tried tweaking that setting! Thank you!

gfairchild, 2021-02-23:
One more minor question: I thought I could put a hostname in the `NUT host` field, but it doesn't seem to work. If I specify the IP of my NAS, I can connect no problem, but if I put the hostname, I get this error:

`[Id 8: 2/23/2021 12:04:23 PM] Connection to Nut Host <hostname>:3493 Failed: Access is denied.`

Is this expected behavior?

gbakeman, 2021-02-23:
Just want to chime in here: I was having issues connecting to my Synology just like you, and the reason seemed to be because I was using an IPv4/v6 combined LAN and the allowed hosts list is very strict about what address you put in. So if I put the hostname in, it resolves to the IPv6 address when I had only put my v4 address in the allowed hosts list.

To work around this, you'll want to make sure you have both IPv4 and v6 addresses in the allowed hosts list, or enter the IPv4 address when you connect. As a note to us developing WinNUT, perhaps we can attempt connection over both addresses if we detect a Synology host or get access denied the first time?

gawindx, 2021-02-23:
Well seen for the IPV4 / 6 combo.

@gbakeman, For your suggestion, it would have to be studied but on this blow I could not certainly not develop it alone, I will need one or more beta testers who have a synology.

It will be seen for v2.1 when it starts to be sufficiently developed.

gfairchild, 2021-02-23:
> Just want to chime in here: I was having issues connecting to my Synology just like you, and the reason seemed to be because I was using an IPv4/v6 combined LAN and the allowed hosts list is very [...]

Nice to note. I just put my local IPv6 address in the list of permitted devices:

[image]

But this still doesn't seem to enable me to connect using the NAS' hostname. WinNut still yields this error:

`[Id 27: 2/23/2021 1:50:36 PM] Connection to Nut Host <hostname>:3493 Failed: Access is denied.`

@gawindx, I'm happy to help test.

gbakeman, 2021-02-23:
Just to double check, are you clicking "Ok" and "Apply" after that? I was only doing one, and that caught me up!

gfairchild, 2021-02-23:
> 
> 
> Just to double check, are you clicking "Ok" and "Apply" after that? I was only doing one, and that caught me up!
> 
> Edit: To confirm, you're adding the IP of your computer running WinNUT and not the NAS, correct?

Yes, correct on both accounts. The 2 IPs in that screenshot are of my PC running WinNUT, and I did apply those settings.

gawindx, 2021-02-23:
I say maybe a stupidity but there is no limit of the number of connections? a session lifetime?

Normally Winnut forces a clean logout but can synology keep an active session for a while?

gbakeman, 2021-02-23:
I just double checked on my end, and I think I may have misremembered what I did! I was just now able to authenticate with an IPv6 connection. I admit I still don't understand how IPv6 works, but for some reason my PC has multiple IPv6 addresses. I opened a connection with a terminal (KiTTY), then opened a network monitor (Process Hacker) and found the exact local endpoint it was using. It ended up being a different address that I thought. I added that to the list of authorized clients, and I can connect now.

Lesson learned here: I guess you either need to figure out exactly which IPv6 address the application is using on your PC to establish the connection, add every IPv6 address assigned to your PC to the authorized clients list, disable IPv6 wholesale on your diskstation, or explicitly use IPv4 when connecting.

gfairchild, 2021-02-23:
> 
> 
> I say maybe a stupidity but there is no limit of the number of connections? a session lifetime?
> 
> Normally Winnut forces a clean logout but can synology keep an active session for a while?

Good question. I can't imagine there is a session limit that's impacting us (i.e., if there is a limit, I'd sure hope it's greater than 2!).

gfairchild, 2021-02-23:
> 
> 
> I just double checked on my end, and I think I may have misremembered what I did! I was just now able to authenticate with an IPv6 connection. I admit I still don't understand how IPv6 works, but for some reason my PC has multiple IPv6 addresses. I opened a connection with a terminal (KiTTY), then opened a network monitor (Process Hacker) and found the exact local endpoint it was using. It ended up being a different address than I thought. I added that to the list of authorized clients, and I can connect now.
> 
> Lesson learned here: I guess you either need to figure out exactly which IPv6 address the application is using on your PC to establish the connection, add every IPv6 address assigned to your PC to the authorized clients list, disable IPv6 wholesale on your diskstation, or explicitly use IPv4 when connecting.

Ah, your comment really helped! After digging around, for reasons I don't understand yet, I see the same thing. Here's a screenshot of my network adapter properties:

[image]

Of particular note is that temporary IPv6 address, which is different than the (non-temporary?) IPv6 address listed above it. I thought to check my Synology logs to see what IP Synology's logs are capturing, and I see this:

[image]

So what Synology is seeing is my temporary IPv6 address, not the other one. When I pasted that IPv6 address into the list of IPs that are permitted to access the NUT service, all of a sudden I can specify the `NUT host` in WinNUT to be either the IPv4, IPv6, or hostname of my NAS; all 3 work equivalently now.

The fact that my NAS is only seeing the temporary IPv6 is concerning because if I'm supposed to rely on a whitelist in the Synology UI and that IPv6 address is changing with some regularity, then that'll mean that the whitelist is functionally useless.

My takeaway here is to stick with IPv4 addresses in the Synology UI as well as WinNUT preferences. It would be really neat if WinNUT tried connecting to both IPv4 and IPv6 addresses when a hostname is provided (since that's one less static IP that needs to be tracked), but I can use the IPv4 address until that gets implemented.

BrckedmyRMpro, 2021-04-01:
Hi Guys,
I have a Synology 214+Play and have just set up a new APC UPS connected to the NAS.  I set up the Power options in the NAS to as shown above and installed the client using the WinNUT-Setup.msi  But after configuring the settings as shown above in the NUT I'm unable to get a connection to the NAS host.

When trying to connect I get the following Error message: [Id 19: 01/04/2021 19:13:17] Connection to Nut Host nutserver host:3493 Failed: Connection Error

[image]

Does anyone have any ides what to try next?  I'm not using IPv6 at all so when I use config/all I only have IPv4 addresses shown for the PC.

Thanx for any help.

gawindx, 2021-04-01:
Did you authorize your ip in the nas config?
[see here](https://github.com/gawindx/WinNUT-Client/issues/47#issuecomment-783869083)

BrckedmyRMpro, 2021-04-01:
Yes I did specify my computer IPv4 in the Synalogy Permitted discstation settings and I can see the ups information in the Nas  ups device information.

gfairchild, 2021-04-01:
Can you try using `monuser` as the username? `upsmon` didn't work for me, but `monuser` did.

BrckedmyRMpro, 2021-04-01:
Thank you for the suggestion, Yes I had already tried monuser but still the same result.

gbakeman, 2021-04-01:
Can you provide the full logs from WinNUT when it tries to connect?

BrckedmyRMpro, 2021-04-01:
Here is the Debug log:

01/04/2021 20:32:23 Pid: 14020 WinNUT : Windows standby, WinNUT will disconnect
01/04/2021 20:48:24 Pid: 14020 WinNUT : Restarting WinNUT after waking up from Windows
01/04/2021 20:49:16 Pid: 14020 Pref_Gui : Pref_Gui Params Saved
01/04/2021 20:49:16 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:49:16 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:49:19 Pid: 14020 WinNUT : Force Reconnect from menu
01/04/2021 20:49:40 Pid: 14020 WinNUT : Connect To Nut Server
01/04/2021 20:50:01 Pid: 14020 WinNUT : Error When Connection to Nut Host.
01/04/2021 20:50:01 Pid: 14020 String : New Log to CB_Current Log : Connection to Nut Host 192.168.100.4:3493 Failed: Connection Error
01/04/2021 20:50:02 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:50:02 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:50:17 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:50:17 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:50:21 Pid: 14020 WinNUT : Open About Gui From Menu
01/04/2021 20:50:22 Pid: 14020 About_Gui : Load About Gui
01/04/2021 20:50:22 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:50:22 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:50:30 Pid: 14020 About_Gui : Close About Gui
01/04/2021 20:50:30 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:50:30 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:50:44 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:50:44 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:50:52 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:50:52 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:51:24 Pid: 14020 WinNUT : Open Pref Gui From Menu
01/04/2021 20:51:24 Pid: 14020 Pref_Gui : Load Pref Gui
01/04/2021 20:51:24 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:51:24 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:51:24 Pid: 14020 Pref_Gui : Pref Gui Opened.
01/04/2021 20:51:39 Pid: 14020 Pref_Gui : Show LogFile

gbakeman, 2021-04-01:
Ok. It looks like we need to make the error messages more clear. :) Just to be sure, when you added your computer's IP address to the authorized devices, did you hit OK then Apply after? I forgot to click Apply one time! If you did that, then it may be worth rebooting your Synology since that has also cleared out some strange NUT server bugs for me.

BrckedmyRMpro, 2021-04-01:
Here is the log after changing the user name to monuser:

01/04/2021 20:51:24 Pid: 14020 WinNUT : Open Pref Gui From Menu
01/04/2021 20:51:24 Pid: 14020 Pref_Gui : Load Pref Gui
01/04/2021 20:51:24 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:51:24 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:51:24 Pid: 14020 Pref_Gui : Pref Gui Opened.
01/04/2021 20:51:39 Pid: 14020 Pref_Gui : Show LogFile
01/04/2021 20:53:02 Pid: 14020 Pref_Gui : Save Parameters.
01/04/2021 20:53:03 Pid: 14020 Pref_Gui : LogFile Enabled.
01/04/2021 20:53:03 Pid: 14020 Pref_Gui : Pref_Gui Params Saved
01/04/2021 20:53:23 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:53:23 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:53:27 Pid: 14020 WinNUT : Force Reconnect from menu
01/04/2021 20:53:48 Pid: 14020 WinNUT : Connect To Nut Server
01/04/2021 20:54:09 Pid: 14020 WinNUT : Error When Connection to Nut Host.
01/04/2021 20:54:09 Pid: 14020 String : New Log to CB_Current Log : Connection to Nut Host 192.168.100.4:3493 Failed: Connection Error
01/04/2021 20:54:13 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:54:13 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:54:15 Pid: 14020 WinNUT : Main Gui Has Focus
01/04/2021 20:54:15 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:54:17 Pid: 14020 WinNUT : Force Reconnect from menu
01/04/2021 20:54:38 Pid: 14020 WinNUT : Connect To Nut Server
01/04/2021 20:54:59 Pid: 14020 WinNUT : Error When Connection to Nut Host.
01/04/2021 20:54:59 Pid: 14020 String : New Log to CB_Current Log : Connection to Nut Host 192.168.100.4:3493 Failed: Connection Error
01/04/2021 20:55:03 Pid: 14020 WinNUT : Open Pref Gui From Menu
01/04/2021 20:55:03 Pid: 14020 Pref_Gui : Load Pref Gui
01/04/2021 20:55:03 Pid: 14020 WinNUT : Main Gui Lose Focus
01/04/2021 20:55:03 Pid: 14020 WinNUT : Update Icon
01/04/2021 20:55:03 Pid: 14020 Pref_Gui : Pref Gui Opened.
01/04/2021 20:55:14 Pid: 14020 Pref_Gui : Show LogFile

BrckedmyRMpro, 2021-04-01:
Yes the Computer IP address is shown in the Permitted devices when I go back in to have a look. I'm just restarting the NAS to check if that does anything.

BrckedmyRMpro, 2021-04-01:
Hahaha!  yes the reboot of the NAS has cleared the error!

[Id 8: 01/04/2021 21:01:09] Connection to Nut Host 192.168.100.4:3493 Established

Thank you and possibly add that as the final set-up instruction.

gbakeman, 2021-04-01:
Thank you for verifying that and getting back to us! I'll see if I can write a wiki page for Synology devices.

BrckedmyRMpro, 2021-04-01:
what does "shut down on NUT's FSD signal" mean?

How can I test that the PC will shut down with a quick test without having to wait for the UPS battery to discharge?

[image]

gbakeman, 2021-04-01:
NUT servers will send an FSD signal when the conditions are met to require shutting down. According to [this guide](https://dan.langille.org/2020/09/10/nut-testing-the-shutdown-mechanism/), you may be able to simulate an FSD by logging in to your Synology with a terminal. I haven't tried this yet though.

gbakeman, 2021-04-01:
Update: I've created a Wiki page [here](https://github.com/gbakeman/WinNUT-Client/wiki/Synology-Devices) in my fork of WinNUT. @gawindx I wasn't able to access the wiki on your project, do you think you could import my page?

BrckedmyRMpro, 2021-04-01:
OK I ticked that option and on the NAS set the shutdown option to 60 seconds.  Then switched off the power to the UPS and within a minute the NAS started to shut down and then the PC.  One observation on powering up the UPS the NAS resumes but takes some time, where as the PC started up faster but the client did not reconnect to the NAS and even though I have the "Re-established connection" in the settings ticked, the client does not seem to be able to reconnect.

Even forcing a manual reconnect does not seem to work.......also seeing the (not responding) quite often with the client app.

gbakeman, 2021-04-01:
That's some great feedback, thank you for testing. Could you collect that info in a separate bug report for us? I think that would help us get it fixed.

BrckedmyRMpro, 2021-04-01:
Still unabl;e to reconnect after the first successful UPS shut down of the NAS and PC...... here is the Debug log:

01/04/2021 21:45:59 Pid: 9544 WinNUT : Force Reconnect from menu
01/04/2021 21:46:20 Pid: 9544 WinNUT : Connect To Nut Server
01/04/2021 21:46:41 Pid: 9544 WinNUT : Error When Connection to Nut Host.
01/04/2021 21:46:41 Pid: 9544 String : New Log to CB_Current Log : Connection to Nut Host 192.168.100.4:3493 Failed: Connection Error
01/04/2021 21:46:41 Pid: 9544 WinNUT : Open Pref Gui From Menu
01/04/2021 21:46:41 Pid: 9544 Pref_Gui : Load Pref Gui
01/04/2021 21:46:41 Pid: 9544 WinNUT : Main Gui Lose Focus
01/04/2021 21:46:41 Pid: 9544 WinNUT : Update Icon
01/04/2021 21:46:41 Pid: 9544 Pref_Gui : Pref Gui Opened.
01/04/2021 21:46:45 Pid: 9544 Pref_Gui : Show LogFile

gawindx, 2021-04-01:
@gbakeman I'll see this for the wiki, I'll have to see if I can give you rights to it.
sorry not to have followed the discussion too much but I am not going through a nas so I don't think I will have been of much help.
and it's time for me to go to bed if I want to be awake enough to go to work tomorrow.

gawindx, 2021-04-01:
@BrckedmyRMpro, if you were able to connect it is because your connection parameters are correct so either it is your nas which is capricious, or it is in winnut that it gets stuck.
have you tried a telnet connection to see what's going on?
I believe I had explained the procedure [here](https://github.com/gawindx/WinNUT-Client/issues/41#issuecomment-702836685).

for identifiers it is:
LOGIN your_login
PASSWORD your_password

if even like that you can't connect, it's because your nas is capricious

gbakeman, 2021-04-01:
Like gawindx said, you can try manually logging in and see what happens. You may also want to reboot the NUT server, with [sudo synoservice --restart ups-usb](url).

Bagu, 2021-06-07:
It seem there is a problem if using DSM 6.2.2-24922
https://www.cachem.fr/ups-domoticz-synology/

Because i also try to set up winnut without success

1. I set ups as name
2. upsmon as user (also try monsuer and an admin user of the nas)
3. secret as password (also try the real password of the admin user above)
4. polling intervall is greater than 1

And the connection is refused

gbakeman, 2021-06-08:
A few things to check -

1. Is your computer added as an authorized machine in the UPS configuration?
2. Are you using IPv6 and IPv4 in your network? Make sure you add all possible host addresses to the authenticated list, or only use IPv4.

Bagu, 2021-06-08:
1-Yes
2-Only ipv4

But, i open a ticket to synology. And the solution is really simple. I just unactivate ups on the Synology Nas, apply, wait few seconds, the reactivate ups on the nas. And everything work like a charm.
Thanks

MimirRnD, 2021-06-30:
> 
> 
> I've just recently set up my UPS (APC Back-UPS Pro) to secure my Diskstations. I've set up UPS monitoring with automated shutdown (to safe mode) in my DS220+ as a server and DS213+ as a client. Now I was looking for a way to include my PCs into this and found this project. Great - love it !!
> 
> Actually it took my only one hour to set everything up including configuring Windows NUT client. After some trial & error I did some research and found that only
> 
> **ups** works as a parameter for the _UPS Name_
> **3493** as _NUT Port_, **upsmon** as _Login_ and **secret** as _password_ are still the default values in DSM6.x

This works perfect
Follow these usernames and secrets
used this on my DS1517+

dwcouch, 2021-07-21:
> Just want to chime in here: I was having issues connecting to my Synology just like you, and the reason seemed to be because I was using an IPv4/v6 combined LAN and the allowed hosts list is very [...]

______
About that IPV6 issue....
I've been  tweaking my network, UPS, PC, etc. And while I had previously set-up WinNUT successfully and it has saved my PC a couple of power outages already - I thought I'd see if I could tweak the reconnection interval to accommodate a reboot of the NAS.  So I changed the interval to 3600, [Apply]'d and then got the following error: 

> 7/21/2021 5:15:04 PM Pid: 11584 UPS_Network : Error When Connection to Nut Host : No connection could be made because the target machine actively refused it 192.168.102.15:3493

After reviewing the thread here I did an 'ipconfig' from a 'cmd' window and found my IPv6 address. Added that to the NAS 's Permitted Devices and then 'Reconnect' in WinNUT. It connected immediately. 

While configuring my network lately it occurred to me I'll never need to use IPV6 INSIDE my network. I'll let my Router and ISP hash that out. So I disabled IPV6 under the Network Adapter properties in Windows. As soon as I did WinNUT disconnected. And it would not reconnect. Even though I had both IPv4 and IPv6 set in the NAS. It did reconnect after I removed the IPv6 IP from the NAS settings.

______

Just wanted to share in case this helps anybody else - including development of a solution for IPv4&6

Now I'm back to my original dilemma... If I reboot my Synolgy DS920+ I'll always have to reconnect WinNUT manually as the current maximum window for reconnection is 30 hours (3600 seconds = 1 hour x 30 tries). So If I reboot my NAS more than 30 hours after the Windows machine I'll have to manually reconnect WinNUT. 

** Feature Request: Ping NAS indefinitely. When it's found, reconnect. 
*** Feel free to throw my name in the hat of Synology Beta Testers too.

Thank you for this little gem. I loose power in my neighborhood all the time. The last big one was while I was re-calculating my watts for a new UPS. OUCH!

Windows 10 PC, Synology DS920+, APC Back-UPS 1500

gbakeman, 2021-07-21:
Thank you for posting that! So just to be clear, when you disabled IPv6 while using it, WinNUT was no longer allowed to connect to the NAS until you removed the IPv6 address in the allowed hosts list?

dwcouch, 2021-07-22:
Glad to provide feedback.

Yes. WinNUT disconnected after I disabled IPv6 on the network adapter and
did not reconnect until I removed the IPv6 address from allowed hosts on
the NAS.

I'll re-confirm with a repeat performance tomorrow morning.

Later after my last post I noted the NAS had a logged in user on the IPv6
IP for another service (though I don't know how since I disabled IPv6...
perhaps a cache of some-sort). I wondered if that had any impact. So I'll
double check.

gbakeman, 2021-07-22:
I think I've noticed some strange behavior, at least from the NUT/upsd server on Synology devices. Sometimes rebooting the unit clears strange bugs. I'll look forward to seeing your results though!

dwcouch, 2021-07-22:
Welll... I was unable to reproduce the issue I had yesterday where WinNUT client would not connect until I disabled IPv6 and cleared the IPv6 IP in the NAS.

Here is a series of test cases and results from what I just ran.

> Ethernet IPV6 Enabled
> WinNUT connected on IPv4
> NAS connections: HTTP/HTTPS on IPv4, CIFS on IPv6
> Both IPv4 and IPv6 are listed in NAS Permitted Devices
> 
> TEST1
> Ethernet: Disabled IPv6
> WinNUT: Manual Reconnect OK on IPv4
> WinNUT: Changed Polling Interval. Reconnect OK on IPv4
> 
> TEST2
> Ethernet: Enabled IPv6
> WinNUT: Manual Reconnect OK on IPv4
> WinNUT: Changed Polling Interval. Reconnect OK on IPv4
> 
> TEST3
> NAS: Removed IPv6
> Ethernet: Disabled IPv6
> WinNUT: Disconnected. 
> WinNUT log: [Id 20: 7/22/2021 12:25:38 PM] Connection to Nut Host 192.168.102.15:3493 Failed: Value cannot be null.
> Parameter name: value
> WinNUT: Auto Reconnect OK on IPv4
> 
> TEST4
> Ethernet: Enabled IPv6
> NAS: Added IPv6
> WinNUT: Disconnected.
> WinNUT log: [Id 26: 7/22/2021 12:31:09 PM] Connection to Nut Host 192.168.102.15:3493 Failed: Value cannot be null.
> Parameter name: value
> WinNUT: Auto Reconnect OK on IPv4
> 
> TEST5
> NAS: Killed CIFS connection. Reconnected CIFS on IPv6
> Ethernet: Disabled IPv6
> WinNUT: Changed Polling Interval. Reconnect OK on IPv4
> NAS: Removed IPv6
> WinNUT: Disconnected.
> WinNUT log: [Id 36: 7/22/2021 12:36:42 PM] Connection to Nut Host 192.168.102.15:3493 Failed: Value cannot be null.
> Parameter name: value
> WinNUT: Auto Reconnect OK on IPv4

I imagine it might be of value to restart the NAS with the PC using IPv6 as well, but the NAS is in the middle of a storage expansion. Let me know if you'd like me to explore that - or anything else.

gbakeman, 2021-07-22:
This is incredibly detailed and useful information, thank you for taking the time. It looks like you've hit a bug in our (re)connection login in tests 3/4/5.

Just so I understand your core concern at the moment - WinNUT did not automatically reconnect when it lost connection (i.e from modifying network adapter properties)? I don't quite understand what you mean by the maximum window for reconnection being too long - are we not trying to reconnect fast enough? I admit it's been awhile since I looked at the code so I can't remember how we handle reconnecting or what the valid range is for retry timeouts. I definitely agree that we need to take a look at connection logic and be more IPvx-aware, especially when dealing with Synos. One thing on my wishlist (when I can get back to developing) is to take a look at that and maybe revamp it some, so that should take care of that. I'd very much appreciate it if you could run more tests.

dwcouch, 2021-07-23:
Hey @gbakeman I believe the entries 'WinNUT: Disconnected.' were simply a matter of the NAS updating the settings. Clicking [Apply] after removing or adding an address makes the UI do a refresh. I assume it's restarting the UPS Server. So I'd not assume it's a bug - it was likely truly disconnected, at least momentarily. So WinNUT performed perfectly - acknowledging the disconnection and then reconnecting.

What brought me to this post was an issue where WinNUT was not reconnecting to my NAS after I had attempted to tweak the polling interval. But as I've reported - I'm unable to re-create that scenario. All was well after I uncovered the IPvx reports on this thread and played with those. 

My NAS is STILL expanding... It looks like it may not be done until Tuesday next week (It's been running 45 hours and it's at 30%!) I'll be happy to run more tests after it's done.

_(This should probably me moved to another thread about the polling interval and reconnecting...)_
So the initial thing - the polling interval; The default is 5 seconds, the range allowed is 1 to 60 seconds. The maximum number of retries is 30. So I take this to mean that the client will attempt to reconnect for a maximum of 30 minutes before it would require a manual reconnect. 

So it's not that it was reconnecting too slowly. But that it gave up too quickly. At least it was when I had it set to 1 second and I rebooted the NAS - WinNUT reached it's max retries after 30 seconds. And that was my concern. If the NAS goes offline for whatever reason WinNUT has a limited amount of time before it gives up. The 'maximum window' to successfully reconnect now being 30 minutes (30x60 seconds). Thus my recommendation to ping indefinitely perhaps with periodic alerts and then reconnect when a ping is successful.

gbakeman, 2021-07-23:
Understood! I think you're right that that may belong in another, but certainly valid report. A loop of pings should be appropriate. Good luck with the expansion.

dwcouch, 2021-08-18:
I'm back! [Expansion went well - took less time than I projected](https://community.synology.com/enu/forum/1/post/124812?page=1&reply=455597). Thank you.

I've had a few power outages and a Windows Update since I last reported on this issue. This morning I have another instance of WinNUT not connecting until I added a new IPv6 Address. I believe the change is the addition of the Teredo Tunneling Pseudo Interface and yet another IPv6. 

> **Issue Report 8/18/2021**
> Windows 10 Pro 
> Version: 21H1 
> Build: 19043.1165
> Windows NUT Client Version: 2.0.7722.30975

**After power outage, restarted PC. WinNUT did not connect.**

CMD >ipconfig shows the following for the Ethernet Adapter:
- Link-Local IPv6 and IPv4
- And the following for Teredo Tunneling Pseudo-Interface:
- IPv6 and the same Link-Local IPv6 - a different IPv6 from the IPV6 of the Ethernet Adapter.

The Synology > Control Panel > UPS [Permitted Synology NAS Devices] Display:
- Ethernet Adapter specified IPv4
- Ethernet Adapter specified IPv6
- Teredo Tunneling specified IPv6

**Added and Applied the Teredo specified Link-Local IPv6** 

**Reconnect success.**

Thoughts: Could there be a way to specify which Ethernet Adapter and IP is used by WinNUT? In this case it seems that the Windows Update added the Teredo TPI and a new IPv6 address. And for whatever reason WinNUT chose that address to use over the other already established IP addresses.

gbakeman, 2021-08-18:
It does sound like we need to make WinNUT more IP-version aware, if anything to make compatibility with Synology more user friendly. I suspect Windows, or at least some component of the .Net framework is deciding what version of IP to use for connecting. I was thinking a good solution would be to attempt connection on one version, then another if the first fails, or let the user specify themselves. I appreciate your work in testing and providing feedback, so now the rest will be up to us. I'm holding up much of any meaningful development at the moment because of a fork I'm working on to implement a new library, so hopefully I'll have that finished soon.

cdkone, 2021-08-20:
Hi all,
I am not sure if this is the right place to ask this question, but I just installed WinNUT 2.1.7740.35837 on my Windows 10 desktop. I am running a Synology DS1512+ with  DSM 6.2.4-25556. The UPS is a Cyberpower BRG1500AVRLCD and is attached via the USB cable.
I was able to get WinNUT to communicated with the Synology UPS Server but the reporting on the main screen is missing all the UPS info and the Remaining Time info. I have checked the Device Info screen and looked at the results from the upsd command and all the info is available. I am attaching screenshots. Can anyone point me in the right direction? 
Thanks
[image]
[image]
[image]

gbakeman, 2021-08-20:
Could you paste a copy of the output from the UPS Variables List?

cdkone, 2021-08-20:
How can I access the UPS Variables List? The selection in the file menu of the WinNUT client is greyed out.

gbakeman, 2021-08-21:
Well, shoot. It looks like we disable the UPS Variables menu item when an unknown UPS is detected (TODO: change this). In the mean time, could you enable logging and post the log file for us, capturing a full connection procedure?

cdkone, 2021-08-21:
Thanks for looking into this, here is the debug log.
[WinNUT-Client-2021-08-21 Debug Log.txt](https://github.com/gawindx/WinNUT-Client/files/7026266/WinNUT-Client-2021-08-21.Debug.Log.txt)

[51 more replies]

closed 2023-07-12 by witzker