gawindx/WinNUT-Client #53: V2.0.5.0 - not a valid msi-installer package

SeeBer61, 2021-01-20:
When trying to install new version 2.0.5.0 (from inside WinNUT client or by running the downloaded setup-xxx.msi file) on W7 the installation fails with an error message saying something like 

"this does not seem like a valid msi-Package go back to ask the developer for a valid one"

Version 2.0.4.0 has installed properly and is running without problems on my system

bernhard

gawindx, 2021-01-20:
so, the problem is simple ...
it is located between the keyboard and the chair, but not on your side, I assure you, it is mine.

The source code is almost finalized but I am waiting for the fix for the simplified Chinese language before deploying the msi package.

Problem is, I forgot that the update process is changelog based, which doesn't seem like such a good idea in the end ...

The error you encounter is therefore normal since WinNUT cannot find the file to download and therefore creates an unusable file.

The installer will be online within a few days, as soon as the Chinese language translation is verified.

I will comment on this issue of the package release.

gawindx, 2021-02-09:
The new version is available, just start the update manually from winNUT and everything should be fine.

I close the issue

closed 2021-02-09 by gawindx