gawindx/WinNUT-Client #59: Two problems !

Dim3333, 2021-02-07:
Hello,

I installed for the first time, WinNUT Client and :
- after re-start of Windows, Winnut don't appears in the systray
- when i click on the executable file, the icon appears in the systray and disappears immediately.

Thanks

gawindx, 2021-02-07:
which version do you have installed? the last stable (2.0.4.0) or the last beta (released a few days ago)?

the last stable version contains bugs on the stability of the connection, bugs fixed in the last beta version (as well as others and some improvements).

Have you already configured the parameters or are they the default ones?

Dim3333, 2021-02-07:
I installed the last stable.
I configured the parameters without problems.

gawindx, 2021-02-07:
have you checked in the taskmgr if the process is present or not? (just to be sure because Winnut is configured for a single instance, it will not run if it is already running)

When you say you can't launch it, is that also valid for manual launch?

Dim3333, 2021-02-07:
Not in the taskmgr. If i click on the icon (.exe), WinNUT appears and disappears immediately.
Yes, i can't launch it manually.

gawindx, 2021-02-07:
So if I am summarizing well, you were able to launch it after installing it and you were able to configure it.

And since then you can no longer execute it.

Can you try to remove the "HKEY_CURRENT_USER\SOFTWARE\WinNUT" key from the registry, that will reset the settings and you should be able to run it again.

It remains to be seen which parameter poses a problem.

On the other hand, I identified a bug on the loading of the default parameters after a new installation so open Winnut once and close it.
Open it a second time and you can configure it.

this bug will be resolved in the next stable version (expected in a few days).

Dim3333, 2021-02-07:
It seems that's OK after delete the "HKEY_CURRENT_USER\SOFTWARE\WinNUT" key from the registry.
The manufacturer, Name, Serial and firmware are indicated as "Unknown".

gawindx, 2021-02-07:
Two possibilities :
- This informations is not retrieved by Nut or the Nut driver
- Bad connection between WinNut and Nut (Possible with version 2.0.4.0) or between NUT and the UPS

Dim3333, 2021-02-07:
I found, for a Synology NAS, the good parameters are : 

NUT Server : IP of NAS
NUT Port : 3493
UPS Name : ups
Delay : 5000
NUT Port : 3493
Login : upsmon
password : secret

In the french traduction : Préférences -> Divers -> Dém**m**arrer avec Windows

Ririx02, 2021-02-09:
"Démmarrer avec Windows (Démarrer)." petite parenthèse en passage => "Démarrer avec Windows (Démarrer)." une petite faute ^^

gawindx, 2021-02-09:
Là, sur celle la, j'ai pas l'excuse de la langue....

C'est corrigé pour la prochaine version, qui sera dispo ce soir.

closed 2021-02-09 by gawindx