gawindx/WinNUT-Client #61: Window Title Not the Same After Open And 'In Focus'

FileCity, 2021-02-09:
WinNut window title is not the same after being just opened and when selected and in focus.
This is an aesthetic issue.
See the two screens below.

[image]
[image]

gawindx, 2021-02-09:
I think I know where it came from.
And personally I prefer to see "Windows NUT Client" than just "WinNUT".

I'll see to change that.

closed 2021-02-20 by gawindx

FileCity, 2021-02-21:
@gawindx, running latest 2.0.7721 and this cosmetic issue is still present.
Upgrade and clean install tested.
Windows title shows WinNUT and changes itself for Windows NUT Client seconds after.

gawindx, 2021-02-21:
Yes, the name "WinNUT" is a translatable string and for some reason that I haven't figured out yet, the window is displayed first with the untranslated china, then it is replaced by its translation.
It doesn't happen with the different controls and I don't know why it behaves this way.

I also don't want to replace the name of the Form in the code because it matches the name of the VB file and the corresponding class.