gawindx/WinNUT-Client #67: Unable to update from 2.0.4.0

UltraHKR, 2021-02-13:
Version is not correctly detected and 2.0.4.0 is still unable to download the correct package.

As shown in the image below;
(The app downloads a zero byte msi package...)

[image]

gawindx, 2021-02-13:
Yes, the reported version number is incorrect.

This is not in itself an anomaly because I have completely updated the update detection process to a much cleaner solution and with a much lighter code.

What I did not anticipate and that the download link no longer works correctly.

You can download the updated version here:
[WinNUT-2.0.7710](https://github.com/gawindx/WinNUT-Client/releases/download/v2.0.7710/WinNUT-Setup.msi)

gawindx, 2021-02-15:
The necessary steps have been taken to repair the links in both the stable version and the development version.

As the new update process is based on Github APIs, this kind of inconvenience should no longer occur.

I close the issue.

closed 2021-02-15 by gawindx