gawindx/WinNUT-Client #68: wrong UPS load with APC Smart UPS 1000, DS218play

Zwobot370, 2021-02-19:
Hi,

with the windows client i got a max load on the "UPS Load" indicator of 125 W, the UPS is able to handle 700 W / 1000 VA.
I have windows 10 and the last version of Winnut (2.0.7710.35902)
DiskStation DS218play is Last firmware DSM 6.2.3-25426 Update 3
APC Smart UPS 1000 (model from 2010) firmware COM 02.1 UPS.05.I (no network card)
[image]

Thank You

Nice Software :)

Zwobot370, 2021-02-19:
The voltage is wrong as well. It should be 240 Volt 50 Hz.

gawindx, 2021-02-20:
When I see the values displayed, I suspect that several values are not provided (or not correctly) and that the defaults are applied.

Which would cause this abnormal power value.

Can you go to "file-> list ups variables", make a copy of the values (there is a button to extract the info) and copy them here.

Afterwards, it will be necessary to see if it is the UPS or the NUT driver which does not provide them.

FileCity, 2021-02-20:
You can also run this command on the NUT server from the command line:
UPSC UPSNAME@LOCALHOST

All variables will be fetched from the driver, then you can troubleshoot.
Refer to this page to find the right driver to use in ups.conf on the NUT server.
https://networkupstools.org/stable-hcl.html

faugusztin, 2021-02-24:
I got a similar issue on my EATON Ellipse ECO 1200.
125W is calculated when there is no ups.realpower.nominal nor ups.current.nominal nor input.voltage (or if your input.voltage is 220V exactly).
The [relevant section of code](https://github.com/gawindx/WinNUT-Client/blob/1e38870300fbe173aa81d6a3d20701fca5dd1a1e/WinNUT_V2/WinNUT_GUI/UPS_Network.vb#L677):

```
Me.OutPower = Double.Parse((GetUPSVar("ups.realpower.nominal", 0)), ciClone)
If Me.OutPower = 0 Then
    Me.InputA = Double.Parse(GetUPSVar("ups.current.nominal", 1), ciClone)
    Me.OutPower = Math.Round(Me.UPS_InputV * 0.95 * Me.UPS_InputA * CosPhi)
Else
    Me.OutPower = Math.Round(Me.UPS_OutPower * (Me.UPS_Load / 100))
End If
```
If ups.realpower.nominal is not defined, then it defaults to 0, in that case we go to If section, not to Else section.
Then it tries to get the current via ups.current.nominal, which is again not present, thus it defaults to 0. It then calculates input voltage (defaults to 220V) * 0.95 * 1 * 0.6 = 125.4W, rounded down to 125W.

A good alternative would be to do ups.power.nominal multiplied to by ups.load, which would give a load in VA. Then we could do the same CosPhi multiplication we do for current to convert the value to watts. While certainly not perfect, it is certainly a better option than fixed 125W.

An example output from my EATON Ellipse ECO 1200:
```
eaton (EATON/Ellipse ECO 1200/01)
battery.charge (Battery charge (percent of full)) : 100
battery.charge.low (Remaining battery level when UPS switches to LB (percent)) : 20
battery.runtime (Battery runtime (seconds)) : 581
battery.type (Battery chemistry) : PbAc
device.mfr (Description unavailable) : EATON
device.model (Description unavailable) : Ellipse
device.serial (Description unavailable) : 000000000
device.type (Description unavailable) : ups
driver.name (Driver name) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 2
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.parameter.vendorid (Description unavailable) : 0463
driver.version (Driver version - NUT release) : 2.7.4
driver.version.data (Description unavailable) : MGE
driver.version.internal (Internal driver version) : 0.41
input.transfer.high (High voltage transfer point (V)) : 264
input.transfer.low (Low voltage transfer point (V)) : 184
outlet.1.desc (Outlet description) : PowerShare
outlet.1.id (Outlet system identifier) : 2
outlet.1.status (Outlet switch status) : on
outlet.1.switchable (Outlet switch ability) : no
outlet.2.desc (Outlet description) : PowerShare
outlet.2.id (Outlet system identifier) : 3
outlet.2.status (Outlet switch status) : on
outlet.2.switchable (Outlet switch ability) : no
outlet.desc (Outlet description) : Main
outlet.id (Outlet system identifier) : 1
outlet.power (Description unavailable) : 25
outlet.switchable (Outlet switch ability) : no
output.frequency.nominal (Nominal output frequency (Hz)) : 50
output.voltage (Output voltage (V)) : 230.0
output.voltage.nominal (Nominal output voltage (V)) : 230
ups.beeper.status (UPS beeper status) : enabled
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 20
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 30
ups.firmware (UPS firmware) : 01
ups.load (Load on UPS (percent of full)) : 41
ups.mfr (UPS manufacturer) : EATON
ups.model (UPS model) : Ellipse
ups.power.nominal (UPS power rating (VA)) : 1200
ups.productid (Product ID for USB devices) : ffff
ups.serial (UPS serial number) : 000000000
ups.status (UPS status) : OL
ups.timer.shutdown (Time before the load will be shutdown (seconds)) : 0
ups.timer.start (Time before the load will be started (seconds)) : 0
ups.vendorid (Vendor ID for USB devices) : 0463
```

faugusztin, 2021-02-25:
I should note that if [reports i found on internet](https://networkupstools.org/ddl/APC/Smart-UPS_1000.html) are true, then APC Smart UPS 1000 doesn't provide the PSU capacity in a NUT variable, thus it is impossible to calculate the power consumption even with my patch above.

If NUT or WinNUT-Client would provide a way to override/add ups.* variables, the user could set their ups.realpower.nominal (in W) or ups.power.nominal (in VA) and then use the provided ups.load to calculate the power consumption.

gawindx, 2021-03-11:
The first beta version of 2.1 has just been released.
It is available [here](https://github.com/gawindx/WinNUT-Client/releases/tag/2.1.7740.35837)

I think the problem you are having should be resolved.
This version is normally perfectly functional.

Venopsis, 2021-03-11:
Hi,

Installed the 2.1.7740.35837 and still having the same full load issue. Also voltage is incorrect as from UPS itself it mentions 230v (live in Belgium) and load should be 10-15%.

faugusztin, 2021-03-11:
As i [mentioned it before](https://github.com/gawindx/WinNUT-Client/issues/68#issuecomment-786252092), the APC Smart-UPS doesn't provide it's own capacity through the Network UPS Tools interface, thus it is impossible for any client software to provide power information. 

To see what variables NUT provides for your UPS, you need to open the File>UPS Variable dialog, press Copy button, then paste the result here. But if the information is not provided, the tool can't show it.

juanantonioadanranera, 2021-12-30:
Hi everyone. I'm testing the Winnut-Client pre-release version 2.1.7740.35837.
My configuration is as follows:
PC: Laptop Lenovo IdeaPad 530S-14IKB.
NAS: Synology DS218+.
UPS: Eaton 3S 700 D.
I've been doing this because the previous (stable) version didn't shown the real UPS load (always at 125 w).
But now, I've lost the value for "Time Remaining", as can be seen in the attached image
Happy new year to everyone.

[image]

gbakeman, 2021-12-30:
Please see issue #95, at some point the remaining time calculation was disabled. I'm not quite sure why, but after the major release coming up I'd like to address it so we have it working again.

juanantonioadanranera, 2021-12-30:
Nice. Let's wait for the next major release. Congratulations for your great software.

gbakeman, 2022-10-02:
FYI for everyone following this: we're tracking the remaining time issue here;

nutdotnet/WinNUT-Client/issues/34