gawindx/WinNUT-Client #69: WinNUT disconnects when UPS running on battery

molnarti, 2021-02-20:
I have recently replaced my UPS with a different model and at the same time updated WinNUT to latest version so not sure what is causing this, but when playing around with the new UPS connected to my server I realized WinNUT looses connection to the nut-server when the UPS is running on battery. Turning on the power makes WinNUT to reconnect instantly. Other on a different computer are able to correctly report the UPS state even when running on battery. Also nut-server is running correctly during the battery operation.

Any idea what could be wrong?

gawindx, 2021-02-21:
This is indeed not normal, there is no reason for winnut to lose connection in this case.

When the connection is lost, is it possible to reconnect Winnut?

Can you activate the logs in debug (in the options) and send them to me?
if possible, delete the log file (always from the options), put your UPS on battery, 10-20seconds after exiting the options (the file will be recreated when exiting the options), let winnut disconnect then try a reconnection.

Only then do you collect the logs to transmit them.

This way we should have as much info as possible.

If winnut is willing to reconnect, UPS on battery, you can also retrieve the list of variables on mains AND on battery history to compare the returned values.

For the list of variables, go to File and List of UPS variables.

molnarti, 2021-02-21:
WinNut does not reconnect. Here's some part from the logs. I thing the issue with the 'Text length must be less than 64 characters long.'
```
WinNUT - 2.0.7706.36312
Reconnection In Progress
Try 0 of 30
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:29Pid: 1476 UPS_Network : Error When Retrieve_UPS_Data : Text length must be less than 64 characters long.
Parameter name: Text
Actual value was WinNUT - 2.0.7706.36312
Connected
On Battery (100%)
Battery OK.
21. 2. 2021 15:51:29Pid: 1476 String : New Log to CB_Current Log : Connection to Nut Host 192.168.50.7:3493 Failed: Text length must be less than 64 characters long.
Parameter name: Text
Actual value was WinNUT - 2.0.7706.36312
Connected
On Battery (100%)
Battery OK.
21. 2. 2021 15:51:29Pid: 1476 UPS_Network : Autoreconnect Enable. Run Autoreconnect Process
21. 2. 2021 15:51:29Pid: 1476 String : New Log to CB_Current Log : Automatic Reconnection
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Nut Server Lost Connection
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Fix All data to null/empty String
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Fix All Dial Data to Min Value/0
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Status Icon Changed
21. 2. 2021 15:51:29Pid: 1476 WinNUT : New Icon Value For Systray : 1344
21. 2. 2021 15:51:29Pid: 1476 WinNUT : New Icon Value For Gui : 1344
21. 2. 2021 15:51:29Pid: 1476 WinNUT : NotifyIcon Text => 
WinNUT - 2.0.7706.36312
Lost Connect To 192.168.50.7:3493
21. 2. 2021 15:51:29Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:48Pid: 1476 WinNUT : Main Gui Lose Focus
21. 2. 2021 15:51:48Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:50Pid: 1476 WinNUT : Main Gui Has Focus
21. 2. 2021 15:51:50Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:50Pid: 1476 WinNUT : Open Pref Gui From Menu
21. 2. 2021 15:51:50Pid: 1476 Pref_Gui : Load Pref Gui
21. 2. 2021 15:51:50Pid: 1476 WinNUT : Main Gui Lose Focus
21. 2. 2021 15:51:50Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:50Pid: 1476 Pref_Gui : Pref Gui displayed.
21. 2. 2021 15:51:54Pid: 1476 WinNUT : NotifyIcon Text => 
WinNUT - 2.0.7706.36312
Reconnection In Progress
Try 1 of 30
21. 2. 2021 15:51:54Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Try Reconnect 1 / 30
21. 2. 2021 15:51:54Pid: 1476 String : New Log to CB_Current Log : Try Reconnect 1 / 30
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Connect To Nut Server
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter AuthLogin
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Process Result With ups.mfr : VAR cyberpower ups.mfr "CPS"
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Process Result With ups.model : VAR cyberpower ups.model "Value2200E"
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Apply Fallback Value when retrieving ups.serial
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Apply Fallback Value when retrieving ups.firmware
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Connection to Nut Host Established
21. 2. 2021 15:51:54Pid: 1476 String : New Log to CB_Current Log : Connection to Nut Host 192.168.50.7:3493 Established
21. 2. 2021 15:51:54Pid: 1476 WinNUT : Update Icon
21. 2. 2021 15:51:54Pid: 1476 WinNUT : NotifyIcon Text => 
WinNUT - 2.0.7706.36312
Connected
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter Retrieve_UPS_Data
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Process Result With battery.charge : VAR cyberpower battery.charge "100"
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Process Result With battery.voltage : VAR cyberpower battery.voltage "25.7"
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Process Result With battery.runtime : VAR cyberpower battery.runtime "6000"
21. 2. 2021 15:51:54Pid: 1476 UPS_Network : Enter GetUPSVar
[... 52 lines cut ...]
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.parameter.vendorid: 0764
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 0
input.transfer.low: 0
input.voltage: 239.0
input.voltage.nominal: 230
output.voltage: 243.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: Value2200E
ups.productid: 0501
ups.realpower.nominal: 1320
ups.status: OB DISCHRG
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

gawindx, 2021-02-21:
Try the latest [version](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.7722.30975)

If I'm not mistaken, I fixed this problem in the next version and from memory, the symptoms you are experiencing may very well be related to this problem as it seems to me that it was indeed crashing the connection to NUT.

molnarti, 2021-02-21:
the latest version fixed it, thanks for the quick solution.

closed 2021-02-21 by molnarti

JoScho20, 2023-07-13:
Hi,
I have the same problem, but with 2.0.7722.30975. As soon as UPS starts running on battery, the connections gets lost. 
I got the software installed in German language.
And in the logfile I find the same "max text length 64 characters" error as mentioned here before.
Following are the logs from the time when I turned power off (so UPS running on battery and connection gets lost):

WinNUT - 2.0
In Verbindung gebracht
Online
Batterie OK
13.07.2023 14:40:14 Pid: 12508 WinNUT : Battery Status => Charging
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter Retrieve_UPS_Data
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Process Result With battery.charge : VAR ups battery.charge "90"
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving battery.voltage
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Process Result With battery.runtime : VAR ups battery.runtime "1620"
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving battery.capacity
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving output.frequency
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving input.frequency
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving input.voltage
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Process Result With output.voltage : VAR ups output.voltage "230.0"
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Process Result With ups.load : VAR ups ups.load "17"
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Process Result With ups.status : VAR ups ups.status "OB"
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving ups.realpower.nominal
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Apply Fallback Value when retrieving ups.current.nominal
13.07.2023 14:40:15 Pid: 12508 WinNUT : UPS is unplugged
13.07.2023 14:40:15 Pid: 12508 WinNUT : Battery Charged
13.07.2023 14:40:15 Pid: 12508 WinNUT : Update Dial
13.07.2023 14:40:15 Pid: 12508 WinNUT : Update Icon
13.07.2023 14:40:15 Pid: 12508 WinNUT : Status Icon Changed
13.07.2023 14:40:15 Pid: 12508 WinNUT : New Icon Value For Systray : 1104
13.07.2023 14:40:15 Pid: 12508 WinNUT : New Icon Value For Gui : 1104
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Deconnect from Nut Host
13.07.2023 14:40:15 Pid: 12508 String : New Log to CB_Current Log : Vom Nut Host getrennt
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Force Disconnect
13.07.2023 14:40:15 Pid: 12508 WinNUT : NotifyIcon Text => 
WinNUT - 2.0
Wiederverbindung läuft
Versuchen Sie 0 von 30
13.07.2023 14:40:15 Pid: 12508 WinNUT : Update Icon
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Error When Retrieve_UPS_Data : Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT - 2.0
In Verbindung gebracht
Batterie (90%)
Batterie OK.
13.07.2023 14:40:15 Pid: 12508 String : New Log to CB_Current Log : Verbindung zum Nut Host 192.168.178.60: 3493 fehlgeschlagen: Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT - 2.0
In Verbindung gebracht
Batterie (90%)
Batterie OK.
13.07.2023 14:40:15 Pid: 12508 UPS_Network : Autoreconnect Enable. Run Autoreconnect Process
13.07.2023 14:40:15 Pid: 12508 String : New Log to CB_Current Log : Automatische Wiederverbindung
13.07.2023 14:40:15 Pid: 12508 WinNUT : Nut Server Lost Connection
13.07.2023 14:40:15 Pid: 12508 WinNUT : Fix All data to null/empty String
13.07.2023 14:40:15 Pid: 12508 WinNUT : Fix All Dial Data to Min Value/0
13.07.2023 14:40:15 Pid: 12508 WinNUT : Status Icon Changed
13.07.2023 14:40:15 Pid: 12508 WinNUT : New Icon Value For Systray : 1344
[... 91 lines cut ...]
13.07.2023 14:40:45 Pid: 12508 WinNUT : NotifyIcon Text => 
WinNUT - 2.0
Wiederverbindung läuft
Versuchen Sie 1 von 30
13.07.2023 14:40:45 Pid: 12508 WinNUT : Update Icon
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Try Reconnect 1 / 30
13.07.2023 14:40:45 Pid: 12508 String : New Log to CB_Current Log : Versuchen Sie erneut, 1 / 30 zu verbinden
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Connect To Nut Server
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Enter AuthLogin
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Process Result With ups.mfr : VAR ups ups.mfr "EATON"
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Process Result With ups.model : VAR ups ups.model "Eaton 3S 850 "
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Process Result With ups.serial : VAR ups ups.serial "Blank"
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Enter GetUPSVar
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Process Result With ups.firmware : VAR ups ups.firmware "02.08.0010"
13.07.2023 14:40:45 Pid: 12508 UPS_Network : Connection to Nut Host Established
13.07.2023 14:40:45 Pid: 12508 String : New Log to CB_Current Log : Verbindung zum Nut Host 192.168.178.60:3493 hergestellt
13.07.2023 14:40:45 Pid: 12508 WinNUT : Update Icon
13.07.2023 14:40:45 Pid: 12508 WinNUT : NotifyIcon Text => 

Any idea how to fix this?
Thanks
Jo

JoScho20, 2023-07-13:
Top. Thanks. Installed latest version from new site and the problem is resolved.