gawindx/WinNUT-Client #71: calibration tab

esofron, 2021-02-23:
About calibration tab in settings.
Does this setting up values on the ups?
Or it just reads the current settings from the ups?

thanks.

FileCity, 2021-02-24:
@esofron, the calibration values are not uploaded to and fetched from your UPS, @gawindx can confrm if necessary.
It's an aesthetic configuration on how you want the gauges to reflect the values according to your specific UPS.
see below my configuration for a 120V UPS with 24 volts batteries.
For 240v and 50 Hz or with a 12v battery UPS, you can adjust those values to have it visually more pleasant.

[image]

gawindx, 2021-02-26:
Sorry for this late response, little time for me these days.

Effectively, the calibration is only for display configuration and has no impact on the NUT server.
It therefore only serves to define the minimum and maximum value of each gauge according to the needs of each one.

Even if I have in project to give the possibility of a semi automatic adjustment which will read and define the values of the interface, this operation is for the moment entirely manual and the default values are the "standard" values which allow correct display of values. Everyone is free to adapt for a more or less precise range or even closer to local values (for example, in France the 230v mains voltage can oscillate between 220 and 250 depending on whether you are in town, in the countryside etc ...) .

closed 2021-03-15 by gawindx