gawindx/WinNUT-Client #74: Problems with the character field length in the de-DE version

ppxp007, 2021-03-05:
Hi there,

everything works fine until I disconnect the 230V plug from the UPS:

Errormessage:
- [Id 10: 04.03.2021 22:10:32] Verbindung zum Nut Host 192.168.xxx.xxx: 3493 fehlgeschlagen: Der Text muss kürzer als 64 Zeichen sein.
- Parametername: Text
- Der tatsächliche Wert war WinNUT - 2.0
- In Verbindung gebracht
- Batterie (80%)
- Batterie OK.

WINnut then cannot connect to the Synology UPS server because a text field value is too large (longer than 64 characters). Can I force English to test this? Do you need any other information?

- UPS (CPS/ValueII_700E/Unknown)
- battery.charge (Description unavailable) : 99
- battery.charge.low (Description unavailable) : 10
- battery.charge.warning (Description unavailable) : 20
- battery.mfr.date (Description unavailable) : CPS
- battery.runtime (Description unavailable) : 5997
- battery.runtime.low (Description unavailable) : 300
- battery.type (Description unavailable) : PbAcid
- battery.voltage (Description unavailable) : 9.3
- battery.voltage.nominal (Description unavailable) : 12
- device.mfr (Description unavailable) : CPS
- device.model (Description unavailable) : ValueII_700E
- device.type (Description unavailable) : ups
- driver.name (Description unavailable) : usbhid-ups
- driver.parameter.pollfreq (Description unavailable) : 30
- driver.parameter.pollinterval (Description unavailable) : 5
- driver.parameter.port (Description unavailable) : auto
- driver.version (Description unavailable) : DSM6-2-25364-191230
- driver.version.data (Description unavailable) : CyberPower
- driver.version.internal (Description unavailable) : 0.38
- input.transfer.high (Description unavailable) : 295
- input.transfer.low (Description unavailable) : 167
- input.voltage (Description unavailable) : 226.0
- input.voltage.nominal (Description unavailable) : 230
- output.voltage (Description unavailable) : 226.0
- ups.beeper.status (Description unavailable) : enabled
- ups.delay.shutdown (Description unavailable) : 20
- ups.delay.start (Description unavailable) : 30
- ups.load (Description unavailable) : 9
- ups.mfr (Description unavailable) : CPS
- ups.model (Description unavailable) : ValueII_700E
- ups.productid (Description unavailable) : 0501
- ups.realpower.nominal (Description unavailable) : 390
- ups.status (Description unavailable) : OL
- ups.test.result (Description unavailable) : No
- ups.timer.shutdown (Description unavailable) : -60
- ups.timer.start (Description unavailable) : -60
- ups.vendorid (Description unavailable) : 0764

Microsoft® .NET Framework Version:
- 2.0.50727.9151
- 3.5.30729.9135
- 4.8.3752.0

thanks :)

gawindx, 2021-03-05:
which version of WinNUT do you have?
If necessary, you can find the exact version in the "about" menu.

ppxp007, 2021-03-05:
[image]

gawindx, 2021-03-05:
ok, this is the last version.
Could you provide me with the debug logs (you have to activate them beforehand in the options and set them to debug level)
This will give me more information on the problematic character chain.

as for forcing it in another language: in the installation directory of WinNUT, you will find a directory containing the localized dlls.
I think removing the dll for german (or renaming it) should force WinNUT to switch to the neutral language, ie English.

By the way, if you find translations that are imprecise or that could be improved, do not hesitate to put it back up because the translation was done with google translate and I do not speak German at all.

ppxp007, 2021-03-05:
Ok done, if I rename or delete the de-DE language folder (or the file it contains), a new folder/file is simply created again when I start the program.

[WinNUT-client-2021-03-05.log](https://github.com/gawindx/WinNUT-Client/files/6090334/WinNUT-client-2021-03-05.log)

gawindx, 2021-03-05:
Try to copy dll from  another directory and replace german dll

ppxp007, 2021-03-05:
The WinNUT-client.resources.dll exist only in DE, FR, RU and CN folders.

gawindx, 2021-03-05:
yes, but if you replace DE translation with another, WinNUT will not find the good translation in the correct directory and it normally fallback to neutral language.

gawindx, 2021-03-05:
Replace only the dll file, don't modify the directory itself.

ppxp007, 2021-03-05:
a bit crazy - but it works without error message

ppxp007, 2021-03-05:
Error narrowed down. Do you still want to fix this? sometime?

gawindx, 2021-03-05:
It will be resolved, no worries about it.

On the other hand it will be for v2.1, it will be necessary to wait a little the time that it is sufficiently mature to be published (it will not take weeks either) and it will be on the dev branch (therefore in beta) in a first step because I bring very big changes on the code in anticipation of future evolutions.

If you don't have too many problems with the English version at the moment, I'll leave the issue open and let you know as soon as the version is published.
There may be a few bugs (I hope not to miss too many) but it should be functional for normal use.

ppxp007, 2021-03-06:
Je suis heureux de lire cela ! Merci pour votre engagement.

traduit avec www.deepl.com/translator :)

gawindx, 2021-03-11:
The first beta version of 2.1 has just been released.
It is available [here](https://github.com/gawindx/WinNUT-Client/releases/tag/2.1.7740.35837)

I think the problem you are having should be resolved.
This version is normally perfectly functional.

gbakeman, 2022-08-07:
To everyone following this, please follow our latest issue at https://github.com/nutdotnet/WinNUT-Client/issues/5, there's a new pre-release that should fix this. Please provide your feedback there.