gawindx/WinNUT-Client #75: Can't start WinNUT 2.1.7740 in Windows 10

lunstudy, 2021-03-16:
Hi guys,

I install WinNUT 2.1.7740 in Windows 10, then start it, the WinNUT show error message as below, how can I fix it?

[image]

gawindx, 2021-03-16:
The installed version is a beta version (pre-release) and therefore for testing purposes.

If WinNUT has never been installed before, I will recommend installing the latest version in 2.0.

I take note of the problem but it would be interesting to know if WinNUT has already been installed before or if it is a new installation (the registry keys are created at the first launch so the problem could come from there given the message error encountered).

gawindx, 2021-03-16:
[latest stable version](https://github.com/gawindx/WinNUT-Client/releases/tag/v2.0.7722.30975)

lunstudy, 2021-03-16:
Yes, this Windows 10 OS is first time run WinNUT, and I also test install the latest version 2.0 in this Windows 10 OS, but I still get same error message.

gawindx, 2021-03-17:
Can you generate an error report ("generate report and close winnut" button) and paste it here to have a little more information (the report is copied both to a file and to the clipboard, so you just have to paste it directly here)

lunstudy, 2021-03-17:
I try it("generate report and close winnut" button), but it didn't past any report message, only can copy & past message as below

Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_client.My.MyApplication.Generate_Button_Click(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gawindx, 2021-03-17:
normally, winnut creates a user data directory in the %userprofile%\AppData\Roaming\WinNUT-Client directory as well as a key in the registry (HKEY_CURRENT_USER\SOFTWARE\WinNUT)
if you do not find these 2 elements, it means that WinNUT crashes even before having initialized these data, and therefore from the very beginning of the startup.
This will be consistent with your error message because it implies that it cannot instantiate the module which takes care of these 2 functions.

quick question, what is the language of your pc? (I wonder about a problem of character that would pose a problem during the treatment)

lunstudy, 2021-03-19:
Hi,

My Windows 10 Pro OS default is English(US).

Thanks.

gawindx, 2021-03-22:
considering the error encountered, and especially the fact that it arrives early in the launch of Winnut, I am not surprised that the key is not created.

Where it bothers me is that I do not see why it crashes.
I took care to work only at the user level to avoid having access to admin rights, and at this stage of the code, we only access a user directory and a user registry key.

Which makes me say that the problem comes from one of these 2 points.

So 2 little questions:
- What is the level of rights of your user? (Admin, user or restricted user)
- Does the name of your user contain any particular characters which could pose problems if not supported in the code?

Regarding the registry, I do not see how your session would not have access to it so I think the problem is not there.

If you agree, I will compile a special version for you which will show you a popup at each step to know which line of code it crashes (unless you have access to visual studio in which case you just have to recover the code to know at which line of code an exception is generated).

manjul968, 2021-04-07:
Hi! I have a similar problem with ver.2.1.7740 and 2.0.7722. Windows Server 2012R2 ru. I have Admin rights with default russian name . The last version that ran was 2.0.4 but it didn't work out the shutdown:) Any ideas?
PS "Generate Report" button not copied data to clipboard
Version 2.1.7740
Exception type: System.TypeInitializationException
Exception message: Инициализатор типа "WinNUT_Client_Common.WinNUT_Params" выдал исключение.
Exception stack trace: 
   в WinNUT_Client_Common.WinNUT_Params.Init_Params()
   в WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   в System.EventHandler.Invoke(Object sender, EventArgs e)
   в System.Windows.Forms.Form.OnLoad(EventArgs e)
   в System.Windows.Forms.Form.OnCreateControl()
   в System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   в System.Windows.Forms.Control.CreateControl()
   в System.Windows.Forms.Control.WmShowWindow(Message& m)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   в System.Windows.Forms.Form.WmShowWindow(Message& m)
   в System.Windows.Forms.Form.WndProc(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
version 2.0.7722
Exception type: System.TypeInitializationException
Exception message: Инициализатор типа "WinNUT_client.WinNUT_Params" выдал исключение.
Exception stack trace: 
   в WinNUT_client.WinNUT_Params.Init_Params()
   в WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   в System.EventHandler.Invoke(Object sender, EventArgs e)
   в System.Windows.Forms.Form.OnLoad(EventArgs e)
   в System.Windows.Forms.Form.OnCreateControl()
   в System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   в System.Windows.Forms.Control.CreateControl()
   в System.Windows.Forms.Control.WmShowWindow(Message& m)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   в System.Windows.Forms.Form.WmShowWindow(Message& m)
   в System.Windows.Forms.Form.WndProc(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

lemppari, 2021-11-11:
I have the same problem. Windows Server 2019 1809. Running and installed as admin. WinNUT won't start. Application directory created, registry key not.

Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_client.WinNUT_Params.Init_Params()
   at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

zagi988, 2022-01-10:
Having the same error here. It was installed and worked great for half a year and then all of a sudden:

`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_client.WinNUT_Params.Init_Params()
   at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

zagi988, 2022-02-22:
Still nothing on this? Or on anything?

gbakeman, 2022-02-22:
We're moving forward on a quick patch in #89. I was supposed to have write access to the repo but found out that the invitations expired, so I sent an email to @gawindx hoping they might resend it. Sorry for the wait!

gbakeman, 2022-10-23:
Everyone following this issue - please see the new issue I've created above, https://github.com/nutdotnet/WinNUT-Client/issues/1. Another person has reported the issue after several patches and I'm going to try and get to the bottom of this. This repository is no longer being updated, but the linked one is.

gbakeman, 2022-10-30:
Attention subscribers to this issue - I believe I have a fix ready. Please see the linked issue above.

FatsWaler, 2023-05-26:
This error has just stated happening to me out of nowhere. I have tried the latest stable edition. Is the any step by step instructions to fix it?

Thanks