gawindx/WinNUT-Client #76: panic

k1ng440, 2021-03-21:
Hello,
 i have misconfigured the interval and now I cant even get to the UI. 

How do I remove the existing config?

https://puu.sh/Hr72y/f015c9abda.png

faugusztin, 2021-03-21:
Check the Windows Registry, the tree under the HKEY_CURRENT_USER\SOFTWARE\WinNUT. You probably mean HKEY_CURRENT_USER\SOFTWARE\WinNUT\Connexion, key Delay to be exact.

gawindx, 2021-03-22:
I did not take the time to answer yesterday because the answer given by @faugusztin was correct.

what bothers me is that you normally wouldn't have been able to indicate a value of zero for the delay.
I would therefore be curious to know how you proceeded to correct this problem.

k1ng440, 2021-03-22:
Hello @gawindx
by default the interval field was 0 for me. I saved the changes and its passed the validation if there is any

thanks

gawindx, 2021-03-22:
ok, this is not normal but that explains your problem.

I do not close the issue, the problem you encountered is clearly a bug, even if I thought I had solved this problem, a point must have escaped me.

thenktor, 2021-06-08:
> 
> 
> Hello @gawindx
> by default the interval field was 0 for me. I saved the changes and its passed the validation if there is any
> 
> thanks

Same thing happened to me.