gawindx/WinNUT-Client #79: Temps de sauvegarde absent

Ririx02, 2021-03-28:
Bonjour,
Je viens de remarqué un petit bug depuis la dernière mise à jour, il n'y a plus le temps restant :
[image]
Hello,
I just noticed a small bug since the last update, there is no more time left

Ririx02, 2021-03-28:
Variable list =>
ups (American Power Conversion/Back-UPS RS 1500G/878.L4 .I)
battery.charge (Description unavailable) : 100
battery.charge.low (Description unavailable) : 10
battery.charge.warning (Description unavailable) : 50
battery.date (Description unavailable) : 2001/09/25
battery.mfr.date (Description unavailable) : 2017/01/19
**battery.runtime (Description unavailable) : 3552**
battery.runtime.low (Description unavailable) : 120
battery.type (Description unavailable) : PbAc
battery.voltage (Description unavailable) : 27.0
battery.voltage.nominal (Description unavailable) : 24.0
device.mfr (Description unavailable) : American Power Conversion
device.model (Description unavailable) : Back-UPS RS 1500G
device.serial (Description unavailable) : 3B1628X19186
device.type (Description unavailable) : ups
driver.name (Description unavailable) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.version (Description unavailable) : DSM6-2-25510-201118
driver.version.data (Description unavailable) : APC HID 0.95
driver.version.internal (Description unavailable) : 0.38
input.sensitivity (Description unavailable) : medium
input.transfer.high (Description unavailable) : 288
input.transfer.low (Description unavailable) : 176
input.transfer.reason (Description unavailable) : input voltage out of range
input.voltage (Description unavailable) : 230.0
input.voltage.nominal (Description unavailable) : 230
ups.beeper.status (Description unavailable) : disabled
ups.delay.shutdown (Description unavailable) : 20
ups.firmware (Description unavailable) : 878.L4 .I
ups.firmware.aux (Description unavailable) : L4
ups.load (Description unavailable) : 42
ups.mfr (Description unavailable) : American Power Conversion
ups.mfr.date (Description unavailable) : 2016/07/15
ups.model (Description unavailable) : Back-UPS RS 1500G
ups.productid (Description unavailable) : 0002
ups.realpower.nominal (Description unavailable) : 865
ups.serial (Description unavailable) : 3B1628X19186
ups.status (Description unavailable) : OL
ups.test.result (Description unavailable) : No test initiated
ups.timer.reboot (Description unavailable) : 0
ups.timer.shutdown (Description unavailable) : -1
ups.vendorid (Description unavailable) : 051d

FileCity, 2021-03-28:
J'ai aussi remarqué ce comportement avec la Pre-release 2.1,

Ririx02, 2021-03-28:
Je suis aussi en V2.1.7740.35837

FileCity, 2021-03-28:
J'utilise la version 2.0 sur les machines critiques pour cette raison.
Je test la 2.1 mais ne l'utilise pas en production.

gawindx, 2021-03-31:
Je rebondis tardivement, j’avais déjà remarqué ce bug dans la version 2.1, mais pas vu pour la 2.0.

Des lignes de codes commentés et je ne les ai pas vu, donc elles étaient restés commentées...

Ririx02, 2021-03-31:
Si c'est juste à dé-commenter, ça ira vite à rectifier, courage à toi (enfin vous, l'équipe de dev, si j'ai bien suivi l'histoire) de nous fournir ce soft 👍

gawindx, 2021-03-31:
Merci
L’équipe et moi même te remercions, c’est à dire moi et... moi (et quelques contributeurs quand même)