gawindx/WinNUT-Client #88: Client disconnected after waking up from sleep

ldellus, 2021-06-25:
Hello,

I'm running a Windows NUT Client version 2.1.7740 on a Windows 10 PC, connecting to a NUT server running on Ubuntu. Everything works fine until my PC wakes up from sleep. The NUT Client doesn't reconnect automatically, I have to reconnect it manually.

Is there a way to configure the client so that it reconnects automatically after waking up?

Thanks,

Laurent

johnrando, 2021-07-09:
I am also experiencing this issue, running Win10 21H1 and WinNUT 2.1.7740.35837.

polarlightning, 2021-08-06:
I think I have related issue, that could be fixed by the same code, as I use anti-virus software, that secures Internet banking sessions by disallowing all other (TCP/IP) socket connections (by default) other than the browser session with the bank.  After I've done my banking (takes more time than the 30s retry period of Windows NUT client) and all communications are again enabled, the Windows NUT client does not recover from the situation and manual reconnect is needed.   Obviously this is not desirable and the "Failure is a standard mode of operation"-principle should rule i.e. bad connectivity should be tolerated and normal operation should be restored as soon as it is sensibly possible.

Hedders, 2021-11-05:
Fascinatingly, this doesn't seem to happen in Windows 11. Since upgrading my PC to Windows 11, WinNUT reconnects to the master on my Synology NAS a few seconds after waking from sleep.