gawindx/WinNUT-Client #89: Hibernate Type Of Stop doesn't hibernate computer

robotics13, 2021-07-08:
When I configure WinNUT to Hibernate my computer the Shutdown GUI will display when the NUT Server sends the FSD signal. The Shutdown GUI will perform the countdown and then hang. When configured to Shutdown instead of hibernate, the shutdown process executes properly.

Steps to reproduce:

1. Configure WinNUT Shutdown Options as follows
* Type of Stop = Hibernate
* Delay to SHutdown (sec) = 15
* Shutdown on Nut's FSD Signal = Checked
2. Disconnect UPS from AC power, wait for NUT Server to send FDS
3. Shutdown GUI will appear, wait for counter to reach zero
4. Shutdown GUI will hang and computer will stay powered on

I suspect this has something to do with the calls to SetSystemPowerState in the Select Cast statement at line 979 in WinNUT.vb. It looks a bit odd passing a Boolean value when the function is declared as expecting an integer. https://www.tek-tips.com/viewthread.cfm?qid=1086465 talks about similar code only passing integers 1 and 0 instead of Booleans. It has been years since I messed with VB.Net so passing a Boolean could be valid, it just looks off.

My computer is running Windows 10 21H1, NUT Server is the built in Synology NUT server, running latest version of WinNUT

robotics13, 2021-07-09:
I took a second look at this and the problem was my computer does not have the Hibernate option enabled.

Not sure if the a try catch around the Hibernate command would be helpful or if an update to the documentation/UI would be better.

g-cranston, 2021-12-01:
I have come all the way to the same final hurdle:  Shutdown works great, Hibernate/Sleep just don't trigger an action, whether at the end of the timer, clicking the immediate shutdown button on the timer, or without timer.  I have hibernate switched and working otherwise.  Have tried this on a PC and laptop, for both v2.0.7722.30975 (stable) and v2.1.7740.35837 (pre-release).  Windows 10 21H2 / 21H1, built-in Synology UPS server.  If power is restored, timer window disappears and it carries on as normal.

I don't know if it has any bearing at all, but on my current windows machines, the command:
`shutdown.exe /h /t 5`  does nothing (shows the syntax guide) when I would expect it to delay hibernation for 5 secs from documentation.

Whereas:
`shutdown.exe /h`  hibernates immediately as it should do.

g-cranston, 2022-02-01:
This worked for me as hoped, once SetSystemPowerState was swapped for SetSuspendState, on Windows 10.  Not sure quality of my coding, but this worked for me:

`    Public Sub Shutdown_Action()
        Select Case WinNUT_Params.Arr_Reg_Key.Item("TypeOfStop")
            Case 0
                Process.Start("C:\WINDOWS\system32\Shutdown.exe", "-f -s -t 0")
            Case 1
                SetSystemPowerState(True, 0)
            Case 2
                SetSystemPowerState(False, 0)
        End Select
    End Sub`

Changed to:

` 
    Declare Function SetSuspendState Lib "PowrProf" (ByVal Hibernate As Integer, ByVal ForceCritical As Integer, ByVal DisableWakeEvent As Integer) As Integer
   
    Public Sub Shutdown_Action()
        Select Case WinNUT_Params.Arr_Reg_Key.Item("TypeOfStop")
            Case 0
                Process.Start("C:\WINDOWS\system32\Shutdown.exe", "-f -s -t 0")
            Case 1
                SetSuspendState(False, False, True)  'Suspend
            Case 2
                SetSuspendState(True, False, True)   'Hibernate
        End Select
    End Sub`

p7654, 2022-02-15:
Hi g-cranson, Is there a fork or installable download that contains this fix?

g-cranston, 2022-02-16:
[https://github.com/g-cranston/WinNUT-Client](https://github.com/g-cranston/WinNUT-Client)

Have made the changes in a fork in case it is of use (plus another fix that works for me).  Apologies if I haven't done this in the appropriate manner (first time).

gbakeman, 2022-02-16:
Hey everyone, again apologies for letting these bugs sit for so long without an update. I've been working for a very long time on a big change, but I'm realizing how important it is that we get these high profile bugs fixed.

I think @gawindx gave me write access to this repo. @g-cranston do you want to make a pull request here with your changes, then I'll pull it in along with a few other waiting PRs and push out an update?

g-cranston, 2022-02-16:
@gbakeman Hopefully I've done the pull request right - seem to be two modifications in one rather than separated.

Thank you for all you and @gawindx have done for this project, a huge and really worthwhile piece of work.  Has really solved my situation where APC's solution didn't!

gbakeman, 2022-02-16:
I'm glad I could be a part of this, and thank you for the PRs. I'm away from the dev computer right now but I'll make it a priority to get your work by next week, hopefully before.

gbakeman, 2022-02-17:
Excellent work by the way, thank you! I just found out that I made a mistake sleeping on @gawindx 's invitation. Apparently they expire in 7 days, and that was a long time ago when they sent it. I'll see if I can contact him (maybe you can too if you want to do more?)

mayziran, 2022-04-20:
> https://github.com/g-cranston/WinNUT-Client
> 
> Have made the changes in a fork in case it is of use (plus another fix for a different issue that works for me). Apologies if I haven't done this in the appropriate manner (first time doing this on github).

I'm having the exact same problems as you, hope these are fixed and there are newer exe

gbakeman, 2022-07-25:
For everyone following this: Thanks to the work of @g-cranston , we have a debug build available. Please download it from [here ](https://github.com/nutdotnet/WinNUT-Client/files/9184971/WinNUTv2.1.8241.26516-debug.zip) and report back results.

DravenSA, 2022-09-28:
I have tried to been using 2.0.7722.30975 and I also can only get it shutdown and not sleep
I updated to 2.1.7740.35837, but that didn't even give me the dialog box when it went below the battery capacity set in settings. 
I tried to run 2.1.8241.26516 as a stand alone, (as it was just in a zip file, so i extracted the folder and ran the  WinNUT-client.exe and again it does the exact same thing at 2.0.7722.30975, where is says it has the delay timer, and then nothing happens.

I can only assume i am to stupid to get this working.
Is there an install file that has a proper fix, as when i see the fixes in the source code, it means nothing to me where to apply them.

I am running NUT Server on a raspberry Pi, and Win Client on windows 10, fully up to date.

It took me the best part of 2 weeks to try and get NUT to run on a pi, so i am not good with code, and I am sure there are other out there like me that need our hands held.

Please can someone help.

Thanks

gbakeman, 2022-09-28:
Hi,

Development of WinNUT has moved to the repository at nutdotnet/WinNUT-Client. This issue can be found duplicated [here](https://github.com/nutdotnet/WinNUT-Client/issues/2). Please try the latest pre-release and see if that works. I'd prefer if all future issues and discussion happen over in the new repository.

Thank you for taking the time to provide your feedback, I hope you'll submit some more!

psychopatt, 2023-06-08:
> Development of WinNUT has moved to the repository at nutdotnet/WinNUT-Client.

This ***really*** should be indicated in the README.  
Hadn't I had this hibernation issue I would never had known since this repo is still the first (and only, it seems) coming up when looking for *winnut* on a search engine.

psychopatt, 2023-06-08:
> This _**really**_ should be indicated in the README.

I created a PR to do so, but I think it would be a good idea to also forbid new issues and PRs if possible.

gbakeman, 2023-06-08:
I wish I had the power to do so! @gawindx granted me access years ago but I didn't accept the invite in time. I've lost all contact with them since and GitHub support hasn't been able to help, so this has sort of become a zombie repository. I might try working on some SEO to boost the new repository up in search rankings though.

psychopatt, 2023-06-08:
Ok, that's too bad. Hopefully the PR will catch the eye of anyone in the same situation and they'll find the new repo more easily.
Anyway, thanks for your work!