gawindx/WinNUT-Client #93: Suddenly won't connect/keeps crashing

katiekat641, 2021-08-13:
WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.1.7740.35837

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 210,
  "MaxInputVoltage": 270,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 210,
  "MaxOutputVoltage": 250,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 6,
  "MaxBattVoltage": 18,
  "MinimizeToTray": "True",
  "MinimizeOnStart": "True",
  "CloseToTray": "True",
  "StartWithWindows": "True",
  "UseLogFile": "False",
  "Log Level": 0,
  "ShutdownLimitBatteryCharge": 30,
  "ShutdownLimitUPSRemainTime": 300,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 0,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "True",
  "DelayBetweenEachVerification": 2,
  "StableOrDevBranch": 0,
  "LastDateVerification": "8/13/2021 1:43:28 PM"
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Object reference not set to an instance of an object.
Exception stack trace: 
   at WinNUT_Client_Common.Nut_Socket.get_IsKnownUPS(String Test_UPSname)
   at WinNUT_Client_Common.UPS_Device.Connect_UPS()
   at WinNUT_Client_Common.UPS_Device.Reconnect_Socket(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : Reconnection Process Started
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket Deconnected
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Nut Server Lost Connection
8/13/2021 1:58:00 PM Pid: 6336 String : New Log to CB_Current Log : Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All data to null/empty String
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All Dial Data to Min Value/0
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update all informations displayed ton empty values
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.1
Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Battery Status => Unknown
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update Icon
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : Reconnection Process Started
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket seems Broken
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket Deconnected
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Nut Server Lost Connection
8/13/2021 1:58:00 PM Pid: 6336 String : New Log to CB_Current Log : Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All data to null/empty String
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All Dial Data to Min Value/0
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update all informations displayed ton empty values
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.1
Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Battery Status => Unknown
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update Icon
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : Reconnection Process Started
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket Deconnected
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Nut Server Lost Connection
8/13/2021 1:58:00 PM Pid: 6336 String : New Log to CB_Current Log : Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All data to null/empty String
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All Dial Data to Min Value/0
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update all informations displayed ton empty values
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.1
Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Battery Status => Unknown
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update Icon
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : Reconnection Process Started
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket seems Broken
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : TCP Socket Deconnected
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Nut Server Lost Connection
8/13/2021 1:58:00 PM Pid: 6336 String : New Log to CB_Current Log : Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All data to null/empty String
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Fix All Dial Data to Min Value/0
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update all informations displayed ton empty values
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.1
Lost Connect To 192.168.0.9:3493
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Battery Status => Unknown
8/13/2021 1:58:00 PM Pid: 6336 WinNUT : Update Icon
8/13/2021 1:58:00 PM Pid: 6336 UPS_Device : Reconnection Process Started
8/13/2021 1:58:30 PM Pid: 6336 WinNUT : NotifyIcon Text => 
WinNUT-Client - 2.1
Reconnection In Progress
Try 3 of 30
8/13/2021 1:58:30 PM Pid: 6336 WinNUT : Update Icon
8/13/2021 1:58:30 PM Pid: 6336 UPS_Device : Try Reconnect 3 / 30
8/13/2021 1:58:30 PM Pid: 6336 String : New Log to CB_Current Log : Try Reconnect 3 / 30
8/13/2021 1:58:30 PM Pid: 6336 UPS_Device : TCP Socket Created
8/13/2021 1:58:30 PM Pid: 6336 UPS_Device : TCP Socket Deconnected
8/13/2021 1:58:30 PM Pid: 6336 UPS_Device : Reconnection Process Started

pparedes1, 2021-08-28:
I'm getting this same error only when I'm connecting using Ethernet. When I unplug and connect using wireless card I receive no error and it connects to the UPS... Not sure what setting might be driving this error.

katiekat641, 2021-08-29:
My issue turned out to be that my IP address had changed. Once I noticed
that and changed it in the settings things went back to normal. Hope that
helps.

gbakeman, 2021-08-29:
Thank you both for the extra information, this should help in fixing the bug. Are you saying that your computer's IP address changed in the middle of a connection?

pparedes1, 2021-08-29:
I think the IP address changing might be the key. Once I read your note I resolved by manually assign a static IP address to my ethernet connection (at the router), release my on my PC (ipconfig /release) and renew the lease and then it no longer had an issue. I suspect we might want to catch that as an exception instead of failing when trying to connect. Appreciate the support and thanks for the work on the app!

gbakeman, 2021-08-29:
We have some related bugs in the network code relating to connections dropping, so hopefully it will be a bird-wth-two-stones kind of thing when we can get to it. In the meantime I've been holding things up with a new library I'm adding that changes a lot of program logic, so I need to get on that (soon) and start testing so we can move on to more the more important issues like this one. Stay tuned and we should have some updates for you!

katiekat641, 2021-08-29:
I believe what had happened was we had

katiekat641, 2021-08-29:
Oops. We had a power outage and I have my computer and NAS connected to my
UPS but not my modem and router. I set up a static IP address later and it
hasn't happened since.