gawindx/WinNUT-Client #98: Cancellation of the extinguishing process

tzeappa, 2021-09-03:
Hello,
I encounter this strange behavior that after the UPS switches to battery and back again to AC the WinNut client keeps popping up and doing "Cancellation of the Extinguishing Process".
This happens usually for days and I have no way of stopping it.
Is this normal and if not, maybe there's something in the settings that I'm doing wrong.
Can you please help me with some info?

Thank you!

gbakeman, 2021-09-04:
That phrase is slightly confusing to me as an English speaker, but the resource is being used in UPS_Network.vb. it looks like the program has detected the UPS coming back online, and is trying to cancel a shutdown. It should only do this once with the UPS comes back, so this doesn't sound like intended behavior. The network code is about to get an overhaul so this bug should be hit then. Thanks for reporting!

eats7, 2021-09-07:
yup having the same issue, and on the newest development branch, it triggers a shutdown event as soon as the app is opened now. even though the battery charge is at 100 percent. I've had to uninstall it as i cant use it having this window pop in and out every second.

RBLink001, 2021-09-18:
> yup having the same issue, and on the newest development branch, it triggers a shutdown event as soon as the app is opened now. even though the battery charge is at 100 percent. I've had to uninstall it as i cant use it having this window pop in and out every second.

I have the same behaviour. The only way to avoid it is to disconnect USB cable from UPS (Synology). After reconnecting the USB cable, it works normally.

tzeappa, 2021-10-01:
I have discovered that rebooting my Synology will resolve the issue, I suppose it's something on the NUT server installed on the nas, not sure what exactly...
I haven't tried disconnecting the USB.

gbakeman, 2021-10-01:
> I have discovered that rebooting my Synology will resolve the issue, I suppose it's something on the NUT server installed on the nas, not sure what exactly... I haven't tried disconnecting the USB.

Have you upgraded to DSM 7? I've had issues with the upsd server misbehaving in DSM 6 requiring either a service restart via command line or rebooting like you said. I remember Synology at least mentioning NUT in their release notes so I wonder if that might help.

tzeappa, 2021-10-02:
> Have you upgraded to DSM 7? I've had issues with the upsd server misbehaving in DSM 6 requiring either a service restart via command line or rebooting like you said. I remember Synology at least [...]

I did, I'm on DSM 7. I tried rebooting the service but the old script doesn't work anymore thus the only option was rebooting.

gbakeman, 2021-10-02:
Ok, good to know, unfortunately. Sorry this is taking so long, I'm going back and forth between developing the underlying library and implementing it into WinNUT (see issue #54 for more info). It's going to take longer than I thought since the last update.

coldfire0200, 2021-11-03:
I just started using WinNUT and notice the issue with Synology Network UPS Server (DS920+, DSM7). This happens when I turn on the vacuum cleaner and surge trips the UPS momentarily. The problem is that, before UPS is tripped, the status is "OL", after UPS enters and leaves battery mode, it status stuck to "FSD OL". Since UPS_Network processes each word in the status string, it first encounters "FSD" however this raises the exception and skips all subsequent words, thus "OL" never has a chance to cancel the shutdown. And the status will be stuck in "FSD OL" until you restart the Synology. 
Also the UPS_Follow_FSD does not seem to control anything, I previously assume it controls handling of FSD but actually not. the FSD is processed regardless.
Two things I have to made to the code as workaround:
`
                            If UPS_Follow_FSD Then
                                LogFile.LogTracing("Stop condition imposed by the NUT server", LogLvl.LOG_NOTICE, Me, WinNUT_Globals.StrLog.Item(AppResxStr.STR_LOG_NUT_FSD))
                                ShutdownStatus = True
                            End If

`
And after processed all word in status string, I would do:
'
                If ShutdownStatus Then
                    RaiseEvent Shutdown_Condition()
                End If
'
The idea is: 
1) only process FSD when flag is set 
2) FSD does not raise exception right away, instead it only raise exception when all words are processed and ShutdownStatus is still true
Item 1 is really not quite necessary but I did it anyway.

g-cranston, 2021-12-01:
I experience this issue also, with a Synology as UPS server, which I assume (for me anyway) is related to [https://community.synology.com/enu/forum/17/post/102712](url)

It seems more like a Synology bug with the information that it passes from its UPS server.  For me, after power is restored and charging again, it sometimes (not always) either continues to indicate in Windows NUT Client "UPS on battery", or returns back to "UPS On Line" for a brief period before then switching again to "UPS on battery".  This status only seems to exist on the NUT clients, not on the actual UPS or NAS info itself.  I assume what is happening is that the UPS server is sending a FSD signal continuously (it doesn't when it goes onto battery initially), and I don't have the option "Shutdown on Nut's FSD Signal" ticked, so the flashing window is the client ignoring that every second (or whatever the polling interval is).

I had been disabling/reenabling the UPS in the Synology control panel, but I note the suggestion of just reinserting the USB cable would be simpler!

shaarkys, 2022-01-20:
Funny, have exactly the same issues, I had even to record slowmotion video to catch this popup - thanks all for the workarounds ! I was almost thinking I have been hacked...

thebatfink, 2022-02-12:
Also have this problem. Server is also DSM 7 synology. On my win 10 client after it boots up, shortly after this starts appearing over and over and it causes a popup to keep flashing up interfering with use of the desktop so I have to kill the program. No power outages or anything, just running the app.