gawindx/WinNUT-Client #100: UPS_Network.AuthLogin does not send LOGIN

dylan-kerr, 2021-09-19:
[AuthLogin](https://github.com/gawindx/WinNUT-Client/blob/master/WinNUT_V2/WinNUT_GUI/UPS_Network.vb#L380) sends the USERNAME and PASSWORD messages, which is sufficient for authentication of subsequent commands, but does not send the LOGIN message. This means that the UPS server does not associate this client with a particular UPS.

In practice, I observed this issue when the UPS goes into the FSD state - the primary/master `upsmon` on the UPS server sees that there are no secondary/slave clients and immediately shuts down the UPS without waiting for WinNUT-Client to notice the FSD flag and shut itself down.

I think I can see how to add the LOGIN step to AuthLogin, but wanted to raise this issue for discussion about how this should be done. From inspection of the `upsd` code this should be a safe change when the UPS server is <https://github.com/networkupstools/nut>, but other servers might have different LOGIN semantics (the [protocol document](https://networkupstools.org/docs/developer-guide.chunked/ar01s09.html) doesn't explicitly specify when LOGIN should or should not be called).
Should calling LOGIN be configurable, perhaps with a checkbox in the server settings? Or can we consider it a core part of the protocol that users shouldn't need to opt in or out of?

Thanks and regards,
Dylan

gbakeman, 2021-09-19:
Thank you for your contribution @dylan-kerr ! I'm sorry you went through the trouble, but we're in the middle of implementing a library that's going to abstract away most of the protocol functionality. Proper LOGIN/LOGOUT support will be one thing that's added. Hopefully we can address your issue soon.

dylan-kerr, 2021-09-19:
Ok, good to know. I'll continue running my fork for now and look forward to seeing the new version!

closed 2021-09-19 by dylan-kerr