gawindx/WinNUT-Client #104: External program start or BAT file.

sRspectrect68, 2021-09-23:
It would be possible to add in the functionalities the possibility to start an external program or launch a BAT file, xxx seconds before the shutdown
tanks, Spectre

gbakeman, 2021-09-23:
This sounds like a great idea, thanks! I think we could have a feature where the user can specify actions that happen when certain events occur.

FileCity, 2021-09-26:
Something very similar has been asked already.
https://github.com/gawindx/WinNUT-Client/issues/39

sRspectrect68, 2021-09-27:
Great...
Can I ask what language WinNut was written in?

gbakeman, 2021-09-27:
If I recall correctly, WinNUT was originally written in AutoIT (or another scripting language), then ported to its current language of Visual Basic .Net.

sRspectrect68, 2021-09-28:
Visual Studio ?

gbakeman, 2021-09-28:
I personally use it, I don't know about @gawindx or anyone else.

ArcticZero, 2021-10-01:
I needed this as well. As a workaround, I set up SSH on my Windows machine, and set my NUT server's upssched script to SSH into Windows and run a batch file and turn off power-intensive processes and set MSI Afterburner to use a low power profile. It's a bit hacky, but hey, it works for now.

gbakeman, 2023-06-11:
Everyone,

This repository is abandoned and no longer developed. I've reopened this issue at the current repository, [here](https://github.com/nutdotnet/WinNUT-Client/issues/92) . Please close this current issue if possible and subscribe/participate in the one I linked.