gawindx/WinNUT-Client #105: Consider moving configuration away from Registry

gbakeman, 2021-09-23:
### Summary
Currently, WinNUT stores its configuration in the Windows Registry. I've seen at least a few people who have had problems with this because of a stale configuration that was left in the registry. It's difficult to clear configuration, view it, save it and restore it.

### Solutions
- Use .Net serialization to store a configuration class as a file somewhere
- Use a common format such as JSON, XML, INI, etc.

hACKrus, 2021-09-26:
This is client for Windows, there registry is common store for settings. If you need INI - use other clients.

FileCity, 2021-09-26:
@hACKrus, I understand your point. Having it in the registry is for convenience.
It's not the easiest location to fetch from to a backup location if a user wants to keep a safe copy in case of a restore or emergency. Having it in on a local file is easier in that case, but...
A menu option can be added to export the configuration from the registry to a file. This file can be JSON or another format.
So it's more a developer personal preference... ... Either locations are fine...
Linux doesn't really have a registry equivalent solution and people are not missing it...
This is my personal opinion.

gbakeman, 2021-09-26:
Those are good points by both, thank you. I'd like to see WinNUT run cross-platform some day so that may necessitate at least abstracting the save system. Most of the work is already done IIRC since we have a NUT_Configuration (or NUT_Settings, I can't remember) object that keeps track of program configuration. All we'd need to do is then make some functionality to save/load it.