gawindx/WinNUT-Client #109: New Branch ?

Anobium, 2021-11-09:
I want to make some updates but no sure whether to create a new branch.

Changes:

1. Remove online update check by adding new Setting option.  The code needs to be standalone to prevent use of internet.
Therefore, hide the online update option from Toast and Setting Tab.  
2. Add new setting to prevent Win10/11 Toast message
3. Add new setting to issue system command xx seconds after UPS issues 'OB' message, cancelling this operation when 'OL' is issued by UPS.
Therefore, add enabling checkbox, time value variable and file select dialog.

Advice plz

Anobium, 2021-11-09:
[image]

New dialog

gbakeman, 2021-11-09:
Thank you for all of the recent suggestions! I'm currently working on a somewhat major overhaul of underlying code so you might want to hold off until then. It's been taking me awhile to get it done, but hopefully I'll have an update ready soon.

Anobium, 2021-11-09:
This is a great project - well done.

----

I am willing to help.  The changes are really improvements on the solution.
I have been able to get them done but I was going to do them again in a new release to get my coding better!

But, if you can add these to the refactored code then I will hold off.

----
### Improvements are

Issue system commend a time period after 'OB' received. Cancelled by 'OL'.
Save and retained screen position.  I would add a option in the setting/misc tab for this.
Reorder Settings/Connection. To be logically correct (no functional change)
Setting to stop Toasts on Win10/11
Setting to hide Update dialogs including Connection/Update tab
For any parameter like Firmware that is NOT supported by the UPS then hide the field in WinNUT form.

gbakeman, 2021-11-10:
I don't think I can take much credit, @gawindx has been the leader on this! :)

I'm having tunnel vision right now and feel like I need to get my integration project done first, but once I have a beta/alpha release out then I'd like to start pulling in changes other people have made. Your changes here are very good and I'd like to get to them sooner rather than later. I'll keep you updated and see what the next steps are once a release is out.

Anobium, 2021-11-11:
> I don't think I can take much credit, @gawindx has been the leader on this! :)
>
> I'm having tunnel vision right now and feel like I need to get my integration project done first, but once I have a [...]

What changes are you making?

gbakeman, 2021-11-11:
Issue #54 is the main issue we're tracking this in - essentially core NUT protocol functionality is being moved into a separate library. You can see my progress on that [here](https://github.com/gbakeman/nut-dot-net), then the branch of WinNUT where I'm integrating it is [here](https://github.com/gbakeman/WinNUT-Client). It's been slow going and I apologize for that, I just hesitate to encourage people to make other changes right now since a lot is probably going to change functionally which might require duplicate work for other enhancements or bug fixes.

blackie333, 2021-12-11:
Could you please try to support NUT emulator/wrapper(on apsupsd) from Martin Machtl, details here: https://www.reddit.com/r/homelab/comments/ar0h9l/apcupsd_nut_wrapper_script_apcupsd_and_nut_on_the/ ? It allows me to use modbus protocol and get LOAD values from my APC SMART UPS.
The connection with v.2.0 is unstable(reconnection loop) and v2.1 gives me error on start.
It's working fine on RPi server and Synology & Home automation plugins clients. 
The last not working client is WinNut which I like very much. Thank you

gbakeman, 2021-12-12:
@blackie333 If their library emulates NUT protocol then WinNUT should be able to communicate without any additional work. Chances are you're experiencing bugs that other people have reported here and will hopefully be fixed soon.

blackie333, 2021-12-13:
OK, thank you. When there will be a new branch/test version of WinNUT available I'd be glad to test.

closed 2022-04-03 by Anobium