gawindx/WinNUT-Client #110: Dev 2.x will not compile - see message

Anobium, 2021-11-09:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	BC30451	'Newtonsoft' is not declared. It may be inaccessible due to its protection level.	WinNUT-client	D:\WinNUT-Client.git\branches\Dev\WinNUT_V2\WInNUT_GUI\Update_Gui.vb	85	Active

```

gbakeman, 2022-04-03:
I noticed you've just closed several issues, did something happen?

Anobium, 2022-04-03:
I had not heard any progress. Assumed the project was on hold.

gbakeman, 2022-04-03:
I haven't made a post about this yet, but I created an organization for .Net NUT projects and will hopefully fork this one to put out a few critical updates. I can add you as maintainer if you'd like.

Either way, please keep your issues open - that lets us track them so we can get to them, eventually :)

Anobium, 2022-04-04:
Good idea. I have reopened one issue so I can track the updates.