gawindx/WinNUT-Client #114: WinNUT startet nicht - WinNUT not stating

philipplindner-media-network, 2021-11-27:
WinNUT Startet nicht und wirft folgende Fehlermeldung aus:
WinNUT does not start and throws the following error message:

`Exception type: System.ArgumentOutOfRangeException
Exception message: Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT-Client - 2.1
In Verbindung gebracht
Online
Batterie OK.
Exception stack trace: 
   bei System.Windows.Forms.NotifyIcon.set_Text(String value)
   bei WinNUT_Client.WinNUT.Event_UpdateNotifyIconStr(String Reason, String Message)
   bei WinNUT_Client.WinNUT.Update_UPS_Data()
   bei WinNUT_Client.WinNUT.Retrieve_UPS_Datas(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

Armin46, 2021-12-12:
Hello i have the same issue. System is Windows 10 64bit.

Exception type: System.ArgumentOutOfRangeException
Exception message: Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT-Client - 2.1
In Verbindung gebracht
Online
Batterie OK.
Exception stack trace: 
   bei System.Windows.Forms.NotifyIcon.set_Text(String value)
   bei WinNUT_Client.WinNUT.Event_UpdateNotifyIconStr(String Reason, String Message)
   bei WinNUT_Client.WinNUT.Update_UPS_Data()
   bei WinNUT_Client.WinNUT.Retrieve_UPS_Datas(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2022-08-07:
To everyone following this issue, please follow our latest issue at https://github.com/nutdotnet/WinNUT-Client/issues/5, there's a new pre-release that should fix this. Please provide your feedback there.