gawindx/WinNUT-Client #119: Doesn't start anymore on Windows 10

zagi988, 2022-01-09:
Here's what happens, doesn't even open the app.

`Exception type: System.TypeInitializationException
Exception message: The type initializer for 'WinNUT_client.WinNUT_Params' threw an exception.
Exception stack trace: 
   at WinNUT_client.WinNUT_Params.Init_Params()
   at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

gbakeman, 2022-01-09:
Appears to be a duplicate of #75

zagi988, 2022-01-09:
Seems so. But my was working for a few months and then out of nowhere - started crashing.