gawindx/WinNUT-Client #120: Minor Issue - Output Voltage reported incorrectly

richard6154, 2022-01-09:
My dashboard reports my output voltage as 262v when its is ~240v. I have check with a power meter 240v is correct output. I'm in Perth Australia so 60hz 240v is nominal for my Cyberpower [CP1300EPFCLCDa].

[image]

gbakeman, 2022-01-09:
Can you please paste the raw output of all variables (found in the File menu)?

richard6154, 2022-01-09:
Is this what you require?

UPS (CPS/CP1500EPFCLCD/Unknown)
battery.charge (Description unavailable) : 100
battery.charge.low (Description unavailable) : 10
battery.charge.warning (Description unavailable) : 20
battery.mfr.date (Description unavailable) : CPS
battery.runtime (Description unavailable) : 2130
battery.runtime.low (Description unavailable) : 300
battery.type (Description unavailable) : PbAcid
battery.voltage (Description unavailable) : 16.0
battery.voltage.nominal (Description unavailable) : 24
device.mfr (Description unavailable) : CPS
device.model (Description unavailable) : CP1500EPFCLCD
device.serial (Description unavailable) : CTLLV2000025
device.type (Description unavailable) : ups
driver.name (Description unavailable) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.version (Description unavailable) : DSM6-2-25510-201118
driver.version.data (Description unavailable) : CyberPower
driver.version.internal (Description unavailable) : 0.38
input.transfer.high (Description unavailable) : 260
input.transfer.low (Description unavailable) : 170
input.voltage (Description unavailable) : 240.0
input.voltage.nominal (Description unavailable) : 230
output.voltage (Description unavailable) : 260.0
ups.beeper.status (Description unavailable) : enabled
ups.delay.shutdown (Description unavailable) : 20
ups.delay.start (Description unavailable) : 30
ups.load (Description unavailable) : 24
ups.mfr (Description unavailable) : CPS
ups.model (Description unavailable) : CP1500EPFCLCD
ups.productid (Description unavailable) : 0501
ups.realpower.nominal (Description unavailable) : 900
ups.serial (Description unavailable) : CTLLV2000025
ups.status (Description unavailable) : OL
ups.test.result (Description unavailable) : No
ups.timer.shutdown (Description unavailable) : -60
ups.timer.start (Description unavailable) : -60
ups.vendorid (Description unavailable) : 0764

gbakeman, 2022-01-09:
Perfect!

Your UPS is indeed reporting 260 volts output. When you measured it, did you take it at the wall or at your UPS? Perhaps a voltmeter inside the UPS is going bad, I would try researching your model to see if other people have the issue.

richard6154, 2022-01-09:
Measured the input from the wall and the output from all 6 UPS points all were 240V. The UPS was new and I thought it maybe faulty so checked thoroughly. The only thing I can see that is inaccurate is that the model number is not reporting the Australian version [missing 'a' on end]

richard6154, 2022-01-09:
The model is also reported as CP1500EPFCLCD by the Diskstation

gbakeman, 2022-01-09:
WinNUT is only able to report what the UPS tells it - like you found, the "A" designation is dropped by the firmware, possibly this is a global version and the manufacturer didn't see fit to reprogram it specifically for the Australian/240v markets. As for the voltage, I can't really say why it's reporting 20 volts off. It's odd because that is also the upper limit when it's supposed to transfer to battery power. Either it just hasn't tripped yet, or it really isn't reading that value and there's a bug somewhere in the firmware. I don't think this is a common issue from what I've seen with some brief research I just did. I would recommend reaching out to the manufacturer to see if they can offer further guidance, potentially a replacement is in order.

JoeNoKnow, 2022-01-15:
I have the same issue of a reported "over" the nominal value of output voltage for my brand new CP1500PFCLCDa, so I doubt it is a hardware issue.  I'm in the US with 120V.  The raw values reported for my unit are:

input.voltage (Description unavailable) : 121.0
input.voltage.nominal (Description unavailable) : 120
output.voltage (Description unavailable) : 137.0

The UPS hasn't been tripped by this value, so no real harm to date.  Would be interested in any solutions offered by either CyberPower or the WinNut-Client team.

gbakeman, 2022-01-17:
Everyone,

Please see [this](https://github.com/networkupstools/nut/issues/581) bug in the NUT repository. It's unclear right now, but it may be a bug in the driver. I'm sorry, but there's not much we can do from the standpoint of WinNUT.

richard6154, 2022-01-17:
Thanks for the followup, glad to see its not your issue. Thanks for WinNut great utility.

flashydave, 2022-01-17:
Subject to the USB VendorId and ProductId I have a fix under [#439](https://github.com/networkupstools/nut/issues/439)  that fixes misreported output voltages. I'm not sure how much CPS share their firmware across products but it may well resolve the problem.
If you can run the driver under debug we might be able to work out whether this is the case.

richard6154, 2022-01-18:
> Subject to the USB VendorId and ProductId I have a fix under [#439](https://github.com/networkupstools/nut/issues/439) that fixes misreported output voltages. I'm not sure how much CPS share their firmware across products but it may well resolve the problem. If you can run the driver under debug we might be able to work out whether this is the case.
> 
> The HID is useful as are the reports for output voltage that hopefully has ID 12 (hex)

richard6154, 2022-01-18:
This isn't a WinNut issue and is well discussed in other threads

closed 2022-01-18 by richard6154