gawindx/WinNUT-Client #121: UPS Load e.g. Watts

H3KT1C, 2022-01-12:
Everything seems to be running smoothly except for hibernate and sleep but that that's another matter.

I've noticed that my UPS Watts load doesn't really seem to change, it always sits on about 135w - 140w no matter what my load is, the other day on the display of the UPS (Eaton 5S 1200) I was using using about 400w and it didn't change within WinNUT.

All that changes is the percentage of usage.

As pictured below in the moment of the screenshot my UPS was stating I was using 242w

[image]

groshreez, 2022-05-04:
This also happens to me with an older APC Smart-UPS 1500. The UPS Load % will go up and down but the Wattage always shows 69W and there's no way that's remotely close to accurate. The Smart-UPS 1500 I have doesn't have a LED display but my previous UPS would show the load up to 400+W. I also don't get any output for remaining battery time, it just shows blank.
[image]

gbakeman, 2022-05-04:
Please see issue #68. Certain UPS's report values like Load incorrectly. Also issue #95, remaining time field was disabled at some point. WinNUT's job is to display information reported by the UPS, so please go to File -> List UPS Variables to verify what information is being reported by the UPS and if WinNUT is displaying it incorrectly.

groshreez, 2022-05-04:
Thanks, I see that my UPS doesn't report ups.realpower.nominal or ups.current.nominal so NUT isn't able to calculate Load Wattage. I'm curious why remaining time field was removed, hopefully there are plans for it to return?!

gbakeman, 2022-05-04:
I haven't been able to answer that myself yet! 😅 I'd like to investigate putting it back in when possible. The WinNUT project is currently in stasis while we either wait for @gawindx or relocate the project over to a GitHub Organization. Stay tuned!

gbakeman, 2022-10-02:
FYI for everyone following this: we're tracking the remaining time issue here;

nutdotnet/WinNUT-Client/issues/34