gawindx/WinNUT-Client #122: Automatic shutdown not requested

alebambi81, 2022-01-24:
Hi, I'm on the latest version and I set the computer to shutdown if the UPS battery gets to 10%.
I don't know why WinNUT automatically shutdown the PC even when the battery is 20 or 25%, it seems that it gets the command from the Synology NAS where it is connected.
The Shutdown on NUT's FSD signal is disabled as you gan see

[image]
 
What could cause the issue?
Thanks

Alessandro - Italy

g-cranston, 2022-02-01:
Could it be doing it on the 'shutdown if runtime lower than (sec)' condition, as it doesn't seem to be reporting a Remaining Time?

What happens if you change that box from 120 to 0?

alebambi81, 2022-02-05:
Hi, I tried with "shutdown if runtime lower than" 0 but I got the same behaviour.
In the bottom there's a message:
"Stop condition imposed by the NUT server" but its checkbox is disabled.

[image]

Alessandro